prusa3d/PrusaSlicer #12765: Configurable First Layer Perimeter and First Layer Infill Speed

secv, 2024-05-31:
I find that first layer issues which are solved by lower speeds generally involve issues with small or pointy and thin perimeters. But there is no setting which affects first layer perimeters only which means infill takes unnecessarily long.

I would like an option to set the first layer speeds for perimeters and infill separately so that troublesome perimeters can be printed slowly and the reliable infill can be printed at whatever speed is appropriate. This way, I can have improved reliability without increasing print time with slow infill.

**Describe the solution you'd like**
I would like for the input field "First Layer Speed" to be split into "First Layer Perimeter Speed" and "First Layer Infill Speed"

Alternatively, the "First Layer Speed" input field would remain as-is with "First Layer Perimeter Speed" and "First Layer Infill Speed" being added below it. Both new fields would default to Zero and inherit the value of "First Layer Speed" unless manually set to a different value, similar to the "Travel" and "Z Travel" fields.

**Describe how it would work**
The first layer feature speeds would function the same as subsequent layers, with perimeters and infill having independently configurable speeds.

**Describe alternatives you've considered**
I tried using a height range Speed modifier to only change perimeter speed for the first layer (0>0.3 height) but this was overridden by the first layer speed.

Related Issues:
#12414 
#10793 
#8607 
#7120 
#1424

neophyl, 2024-05-31:
Have you tried setting the first layer speed to a percentage based setting ?  That way it can use different speeds for the different feature types.
For example first layer set to 50%  - note the different speeds for the inner and external perimeters and the infill.
[image]

secv, 2024-05-31:
I saw that in other issues, but I'm not interested in that workaround since it's proportional to other print settings.

u89djt, 2024-06-01:
How about setting first layer speed to 100% then setting a height range speed modifier? It should be 100% of the modified speed then.

pyro9, 2024-06-01:
I find that first layer problems with small pointy perimeters are often solved by a brim or adding mouse ears instead of slowing down.

closed 2026-09-29 by prusaslicer-bot[bot]