prusa3d/PrusaSlicer #12769: Bridge infill not anchoring to unsupported overhanging perimeters

3dPrinting1127, 2024-06-02:
### Description of the bug

I've been having issues with bridging infill since one of the 2.7.x updates.  Before these updates, I've never had to mess around with the "infill/perimeter overlap" default value (10%). I'm currently having trouble anchoring bridge infill to **unsupported** overhanging perimeters.  I'd like to clarify that the bridge infill properly anchors to external perimeters and supported perimeters just fine using the default 10% infill/perimeter overlap value.  

Here's what I'm printing:
[image]

and here is the result:
[image]

As you can see, bridge infill anchors to external perimeters and supported overhanging perimeters just fine, but it does not anchor to the unsupported segment.  I've tried multiple bridge flow values from 0.7-1 and I increased the infill/perimeter overlap to 20% and it still does not anchor.

**A simple solution to this would be to make the infill/perimeter overlap setting higher, but the consequence of this is that interior infill could be seen from the outside of the print.  I'm really not sure if this is a bug or not, but a much easier fix would be to have a "bridge infill/perimeter overlap" setting that only applies to bridge infill, that way I could set this value to 50%+ while maintaining the infill/perimeter overlap setting to 10%. 

### Project file & How to reproduce

[Overhang.zip](https://github.com/user-attachments/files/15526077/Overhang.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.7.4

### Operating system

Windows 11

### Printer model

MK3S+

tastyratz, 2024-06-08:
Are you using classic or Arachne?
Could it be related to https://github.com/prusa3d/PrusaSlicer/issues/9074 ?

3dPrinting1127, 2024-06-09:
I tried it with both classic and arachne and both generate the same undesired results here.  It doesn't seem to be related to that issue because the perimeters are being printed first and then the bridge infill is printed afterwards as it should be.

LeoB-26, 2024-08-14:
I would reccommend trying to print it 5-10°C hotter. Plastic sticks better when it's hot, and if it's PLA then it'll shrink as it cools so you could also try turning down cooling a smidge. Other than that, you could decrease the extrusion width and/or increase the number of perimeters. I know I've had this happen to me before on my Prusa XL and that did the trick for me.

abferm, 2025-01-06:
Similar/related problem here. The slicer appears to put overlap into the non-bridged portion of a model to anchor the bridge, but when bridging to unsupported overhangs (bridged perimeters in my case), it does not overlap the path. This causes the bridged "infill" not to connect securely to the bridged perimeters.
[image]

My case could probably be corrected by running the bridge infill the same direction as the perimeters, but I'm not seeing an option for that.

tastyratz, 2026-09-03:
This issue is amplified quite a bit by issue described in #9074 

Merging in "No perimeters on bridge areas" from SuperSlicer would likely significantly reduce impact of bridging issues.