prusa3d/PrusaSlicer #12771: Organic Infills

TraylorHead, 2024-06-02:
Would like to see an organic infill. 

I think it would cut down on print times and give an excellent support structure.

u89djt, 2024-06-02:
You might be looking for lightning infill. Here's a quick experiment using organic support as infill for a cube. The material of the lightning infill takes 9 minutes compared with 57 minutes for the organic supports.
[image]
[image]
[cubes with organic and lightning.zip](https://github.com/user-attachments/files/15526629/cubes.with.organic.and.lightning.zip)

TraylorHead, 2024-06-02:
Oh, I'm actually trying the lightning now. The preview looks so oddThanksSent from Samsung Galaxy smartphone.

image.png (view on web)
image.png (view on web)

u89djt, 2024-06-02:
Yeah, the way lightning is started and structured is cunning. The rising sheets are stiffened by adding creases.

closed 2026-09-29 by prusaslicer-bot[bot]