prusa3d/PrusaSlicer #12772: Stuck at 90% Slicing

JamesonRGrieve, 2024-06-02:
### Description of the bug

Attempting to slice the attached file results in hanging at `90% - Generating G-code.` for in excess of half an hour. My specs are 12900k / RTX 3090. Slicing one piece at a time seems to work, attempting to slice two walls at the same time also freezes but I didn't wait half an hour to confirm if it was an equivalent freeze.

All settings work fine on other projects. I tried changing seam position to random as was suggested in another issue and that did not help.

### Project file & How to reproduce

1. Slice the linked file (too large for GitHub).

https://drive.google.com/file/d/1Y0IUOUvYuFyWrZB2OutbXLuR1DtXmTBA/view?usp=sharing

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 11 Pro

### Printer model

MK3S+

phillipintx, 2024-06-03:
Same issue running on MacOS Sonoma Intel i9 32GB. Slicing multi part model for MMU2S

JamesonRGrieve, 2024-06-03:
Confirmed `2.6.1` does not exhibit this behaviour. Trying versions inbetween to isolate problem version.

JamesonRGrieve, 2024-06-03:
`2.7.0`, `2.7.1`, `2.7.2` all confirmed working. `2.7.3` exhibits the same behaviour.

u89djt, 2024-06-03:
I've isolated a tiny offending part of the mesh in the left column, highlighted with move arrows in the screenshot. There's at least one more between there and about 43mm in the other column.
[divide and conquer DnDTiles.zip](https://github.com/user-attachments/files/15537321/divide.and.conquer.DnDTiles.zip)
[image]
Any object labelled NO will cause slicing to hang at 90%. One of the objects is labelled SLOW because while it will slice, it takes a very long time.
The tiny objects you see at the front of the screenshot (the back of the bed) are chunks of the back face of the slice taken from the left tower between 28.5 and 28.9mm. The U shape shows as a faulty mesh, but doesn't stop the slicing. If you activate the tiny NO chunk with any other printable part, it causes the slice to hang at 90%.

phillipintx, 2024-06-03:
@u89djt, Why does prusaslicer NOT have issues generating the gcode for this model prior to 2.7.3?

u89djt, 2024-06-03:
@phillipintx that would appear to be the next question :) I'm another user, so we're both looking from the outside in. I would speculate that the devs have taken a more optimistic approach to trusting the code to cope with the mesh as it's imported somehow, or might just have tweaked a bit of code and introduced an off-by-one error or somesuch. My first suggestion would be to try that tiny sliver in an earlier slicer. If that succeeds, then we only need to care about that chunk. Would you be able to test that? I've only got 2.7.3 installed so far. Happy to install at some point if not.

JamesonRGrieve, 2024-06-03:
I just downloaded the portable release builds and renamed my config folder to prevent config corruption. I can give it a shot after I'm done work.

u89djt, 2024-06-03:
Ah, I forgot we could do that. I've just confirmed that the problem chunk doesn't stop 2.7.2 slicing.
So here's the tiny section taken from the original mesh and a box so that there's something that would require extrusions.
[confounding mesh from DnDTiles.zip](https://github.com/user-attachments/files/15539218/confounding.mesh.from.DnDTiles.zip)
[image]

nealzip, 2024-07-28:
I am having the same issue (Stuck at 90%) with 2.7.4 and 2.8.0 in Mac M1 MacBook 16GB Ram. I'm happy to help troubleshoot and/or upload the file if it helps.

u89djt, 2024-07-29:
Does it slice ok if you do this first?
> Workaround: in prusaslicer right click -> simplify -> extra high -> apply. The result slices quickly.

u89djt, 2024-07-29:
@nealzip Does it slice ok if you do this first?
> Workaround: in prusaslicer right click -> simplify -> extra high -> apply. The result slices quickly.

closed 2026-09-29 by prusaslicer-bot[bot]