prusa3d/PrusaSlicer #12773: MMU painting multi-part object treats entire object as 1 unit, not separate parts

ex-nerd, 2024-06-03:
### Description of the bug

MMU painting of a multi-part object does not allow painting of areas "behind" touching parts. Even if you do this (see example) by moving one part of the way to paint, the slicer ignores any painted area. The two objects in this image are identical, with a both of the V walls fully painted, but as you can see the color stops being propagated at any point where the two parts intersect:

[image]

* There should be a way to hide individual parts while painting multi-part objects so that all walls of all parts can be painted.
* Material painting should respect the color/material on each part in an object even when they are "behind" each other.

Additionally, this bug makes it impossible to paint multiple materials inside of part intersections, e.g. I could paint circles of PETG on a surface that would otherwise be PLA, and create a better bond without having to model those shapes into my design.

My real world example is a lot more complex than this (lots of surface color variation), and this bug results in 2-3x as many perimeter walls and slower print time than I would expect because the algorithm chooses a color that results in a less efficient workflow. You can see the area circled in red here that really only needs 4 total perimeters with some infill in black, but has instead resulted in 4 black and 2 green:

[image]

### Project file & How to reproduce

[multi-part-painting-bug.3mf.zip](https://github.com/user-attachments/files/15528075/multi-part-painting-bug.3mf.zip)

This is a simple example where I have two intersecting cubes. To color the walls, I changed the "part coordinates" of the smaller cube to move it away from the larger one in order to paint the color "behind" it. I then copy/pasted this to create a second copy and moved the smaller cube back into place.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+MacOS-arm64

### Operating system

MacOS 14.5

### Printer model

Prusa XL, 5 tool

ex-nerd, 2024-06-05:
> I imagine it would be prohibitively laborious very quickly, and can't work sometimes

Yeah. Would have to design the 2 parts in CAD with an intersecting "lifter" part to keep the "floating" part in the air. Simple enough with a couple parts, but starts to get messy with anything complex.

I think this really is just an issue where the painting should "belong" to the part rather than to the object.

u89djt, 2024-06-05:
Aha, so I was delighted to find that the lifter part doesn't have to intersect. I just needs to be on the plate, and you can set its perimeters, top, bottom and infill to zero so it doesn't appear in the slice. You can put them anywhere on the plate.

ex-nerd, 2024-06-05:
> Aha, so I was delighted to find that the lifter part doesn't have to intersect.

It doesn't drop/move everything down to zero when you split into objects? E.g. stack two cubes on top of each other and then split. My problem was when I did that, both cubes ended up flat on the build plate. So I was thinking you'd have to design a "stilt" for the upper part and then add a modifier to set perimeter/infill to zero for the area of the stilt.

u89djt, 2024-06-05:
Yeah, have a look in the 3mf file. I've given each part from your original examples a lift body.  If you select the part and not the lift body and drag it upwards in Z, it'll stay there. I'm not at a computer at the moment or I'd build a better example.

ex-nerd, 2024-06-05:
> Yeah, have a look in the 3mf file. I've given each part from your original examples a lift body. If you select the part and not the lift body and drag it upwards in Z, it'll stay there. I'm not at a computer at the moment or I'd build a better example.

Ok, I see what you did. The original multi-part object was split, and then a new multi-part object was created with the un-painted part and a lift body.

I was thinking about how to model the two original bodies in CAD so you can import both at once to ensure they stay locked in the same XYZ space when you split apart to paint. This could be done with a separate "lift body" but the end user would then need to then know the exact Z height to lift the split part back up to, and things get really messy if you bump auto-arrange (I guess PrusaSlicer needs a "group" function separate from the multi-part object function).

u89djt, 2024-06-05:
Yup, just build the entity in CAD with separate bodies, import that into the slicer, add a lift box to the object, then copy and paste that object so there's one for each part. Relocate all the copies of (object plus lift) to the same coordinates, then delete everything except one part and the lift body from each copy.

ex-nerd, 2026-09-18:
The behavior described here is identical in v3. Part of this issue seems to relate to the "default extruder" of a given part, e.g. here the third object (lower left) was set up with red as the part's default extruder with the rest of the part painted blue:

[image]

It still seems to be doing a sort of "pick a point closest to all colors" best fit, which is likely by design, but also explains some of the issues with my real-world example where "inside" the part is the default extruder and painted surfaces only go deep enough to find that mid-point rather than optimize filament usage.

I guess there are enough workarounds, either by modeling changes in directly, or playing with a part's default extruder, that I can close this issue. I still think it would be nice to have a sort of "optimize multicolor filament usage on internal walls" setting, but maybe that can be added someday via the new plugin system in v3.

closed 2026-09-18 by ex-nerd