prusa3d/PrusaSlicer #12774: Auto shutdown

Gramps57, 2024-06-03:
If this is even possible, through code, I would like to have you add to PrusaSlicer an option to automatically shut down the printer after the print finishes.
Thank you.

u89djt, 2024-06-04:
Hi there, I'm a fellow user, not a developer, just agreeing that it would be great if we could power down once a print is finished. I think given that there's a physical switch in charge of the power, the most they could do would be to switch off heaters and the display. I wonder if something like this would be good enough: https://www.amazon.co.uk/Brennenstuhl-Countdown-integrated-adjustable-protection/dp/B09M78B9L5 I haven't tried one, so it's not a recommendation, just a start of a search :) Cheers.

Gramps57, 2024-06-05:
I wondered the same thing (concerning the physical switch) but figured if it could be done with computers and cell phones it might be possible with 3d printers. I appreciate your suggestion, however. 

⁣Thank you, 
Dennis​

u89djt, 2024-06-05:
Ah yeah, there's a parallel to draw. I guess with a desktop PC you have a physical switch governing the provision of voltage rails in the power supply unit, but I think that's contingent on a circuit being closed on the cabling by the motherboard, which isn't a manual switch. Laptops and batteries - I don't know, but I think the momentary power switch provides power to circuitry that closes a circuit in a manner related to that on a desktop motherboard. Phones presumably vaguely similarly. I don't have the live knowledge for this - I'll enjoy googling a bit :)
Best wishes
Dave

closed 2026-09-29 by prusaslicer-bot[bot]