prusa3d/PrusaSlicer #12775: Feature request: Ability to paint straight lines in Multi-material painting tool

ex-nerd, 2024-06-03:
There should be a way an easy way to paint straight lines in the MMU painting tool.

**Is your feature request related to a problem? Please describe.**

This is possibly a variation of https://github.com/prusa3d/PrusaSlicer/issues/5970 which was created before MMU painting was added to PrusaSlicer.

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe how it would work**

There are standard UX tropes for this, which have been described in https://github.com/prusa3d/PrusaSlicer/issues/5970:

- If the user has painted/clicked with the brush tool and then shift-clicks, a straight line would be painted from the last-painted location to the current click location.
  - Changing tools would clear the last-painted location.
  - Additionally, this could be implemented with a preview where the user is presented with an outline of the area that would be painted while the shift key is held down. This would greatly help with alignment and also give visual confirmation that a straight line would be painted (or if nothing is shown, indicating that there is no valid last-painted location to paint from).
- If the user holds down shift while clicking to paint with the brush tool, the cursor would lock to a specific orientation (horizontal or vertical based on the cursor direction).

**Describe alternatives you've considered**

See below.

**Additional context**

In the mean time, a workaround for this is possible using the "Clipping of view" tool to set a boundary. While smart fill and bucket tool ignore the boundary, the regular brush tool does not paint past it. It's tedious, but you can move the clipping boundary around to to create straight lines:

[image]

neophyl, 2024-06-03:
You can also left click and hold the button on a spot on the object and then tap the Alt key.  Then you can click in a different spot.  It will then draw a line between the 2 points (as long as the model geometry is clean between the 2 points).

[image]

ex-nerd, 2024-06-03:
> You can also left click and hold the button on a spot on the object and then tap the Alt key

I can't seem to replicate this with MMU painting (at least, not on a Mac, using the option key)

ex-nerd, 2026-09-02:
This is a duplicate of an older issue already tagged as feature-request here: https://github.com/prusa3d/PrusaSlicer/issues/5970

closed 2026-09-02 by ex-nerd