prusa3d/PrusaSlicer #12776: Enhance Interface Shells

DeanDavis, 2024-06-03:
**Is your feature request related to a problem? Please describe.**
When a single STL model has Embossed Text it is often a use case that the slicer is used to change color at the layer where the embossed text starts.
But because model is a single STL mesh the logic for "Interface Shells" doesn't see this as a spot to provide a 100% infill layer.
If the text is a separate STL that is separate from the base then Interface Shells will slice a 100% infill layer.

**Describe the solution you'd like**
The slicer should consider a layer specified color change the same as if the mesh on and above the color change request is separate from the mesh below and the logic of Interface Shells treat this as two separate "materials" or "volumes".

**Describe how it would work**
Import a single STL. Specify a layer color change. Turn on "Interface Shells". The layer below the color change should be a smooth 100% filled in layer. (What ever the Interface Shell logic is)

**Describe alternatives you've considered**
None.

neophyl, 2024-06-03:
There are already existing issues open here that deal with this.  For example https://github.com/prusa3d/PrusaSlicer/issues/8818 or https://github.com/prusa3d/PrusaSlicer/issues/6734

DeanDavis, 2024-06-03:
True, but my description gives the issue a path to go down on exactly how to implement the fix and the UI for it.
Using Interface Shells option allows those who want this option for things like art objects to turn it on and those who print practical parts to turn it off (the default) if desired.
I didn't even mention that Interface Shells fails to work even when using separate STL objects unless you have assigned different colors to the different STL objects. This is sub-optimal as well as there is a use case to have embossed text on an object and the entire object is of a single color.
As far as I can tell right now, if you wanted a singular colored object with embossed text on top there is no way to have the layer just beneath the text to be solid contiguous infill using any option.

u89djt, 2024-06-04:
Here's a workaround off the top of my head, but it looks like the slicer is treating the extrusions as part of a continuous body - there are no bridges indicated in the text held up in the air by the empty bodies, so it should be fine, or at least amenable to optimization.
[text on contiguous surfaces.zip](https://github.com/user-attachments/files/15545637/text.on.contiguous.surfaces.zip)
[image]
Essentially, set up your text on your object, then copy and paste it (not add an instance), type in the same position coordinates as the original object, then add perimeters and layers and infill settings to the original object (not the text additions) and zero everything.

closed 2026-09-29 by prusaslicer-bot[bot]