prusa3d/PrusaSlicer #12779: Možnost po skončení tisku automaticky vytáhnout fillament

DKotekCZ, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
Když mi běží poslední tisk, tak po dotisknutí mi tiskárna vychladne a filament mi zůstane uvnitř, a ten musím pak buď ustřihnout, anebo zas zahřát (pak navíc musím čekat na vychlazení) a vytáhnout, abych mohl cívku uklidit.

**Describe the solution you'd like**
Buď nastavením v tiskárně, anebo alespoň instrukcí do (b)gcode.

**Describe how it would work**
Buď v tiskárně mít možnost nastavit, aby u probíhající tisku, po jeho dokončení automaticky vysunula filament, anebo to alespoň moc přímo ve sliceru přidat jako instrukci na konec.

**Describe alternatives you've considered**
To je ta varianta s úpravou gcode, ale lepší je to mít přímo v tiskárně, třeba v menu Ladění.

**Additional context**
none

pavolNovak, 2024-06-21:
This behavior is directly made by Prusa in END g-code :
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+10, max_print_height)}{endif} F720 ; Move print head up
G1 X0 Y210 F7200
G1 E2 F5000
G1 E2 F5500
G1 E2 F6000
G1 E-15.0000 F5800
G1 E-20.0000 F5500
G1 E10.0000 F3000
G1 E-10.0000 F3100
G1 E10.0000 F3150
G1 E-10.0000 F3250
G1 E10.0000 F3300

M140 S0 ; turn off heatbed
M107 ; turn off fan
M702 C
G4 ; wait
M221 S100 ; reset flow
M900 K0 ; reset LA
{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}
M104 S0 ; turn off temperature
M84 ; disable motors

Jan-Soustruznik, 2025-05-12:
Ahoj @DKotekCZ, ve Sliceru je možné přidat příkaz pro vysunutí filamentu přímo do bloku **Konec G-code** v záložce **Tiskárny**, sekce **Vlastní G-code**. Pokud se však jedná o požadavek na implementaci této funkce přímo ve firmwaru tiskárny, je potřeba vytvořit požadavek přímo týmu firmware na GitHubu zde: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues 

Tiket uzavírám jako vyřešený.

closed 2025-05-12 by Jan-Soustruznik