prusa3d/PrusaSlicer #12780: Wrong Leveling Temperature with Generic Flex Filament

foorschtbar, 2024-06-05:
### Description of the bug

I sliced a model with the Generic Flex Filament. Selected Extruder 3 in Slicer. Loaded TPU in Extruder 3. I can see that the mesh bet leveling is done with 200°C and the nozzle drips result in very ugly sheet. Checked the gcode, i could see the temperature is 200°C. Normal is 170°C as far as i know.

[image]

Switched back to a PLA/PETG extruder, there is the 170° for leveling back in the gcode.

### Project file & How to reproduce

[Dichtung Ovalleuchte REV Ritter 0090060410 - PS.zip](https://github.com/user-attachments/files/15595809/Dichtung.Ovalleuchte.REV.Ritter.0090060410.-.PS.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4-win64

### Operating system

Windows 11

### Printer model

XL 5T

rtyr, 2024-06-06:
210C is intended, some materials require higher temps for proper load cell measurements. It's rather FW/HW requirement even though currently executed via start gcode.

closed 2024-06-06 by rtyr

foorschtbar, 2024-06-07:
Thank you for your quick reply. 

Is it possible to set THIS temperature manually in the profile or what can I do here? Currently I have to edit the gcode every time. the print is then successful with the change on the two possitons. no dripping and no problems with leveling etc.

rtyr, 2024-06-07:
You can change this in the start g-code `(filament_type[0] == "FLEX") ? 210`, you can then save it as custom printer profile.

rtyr, 2024-06-07:
Or alternatively you can just change filament type in the filament profile.