prusa3d/PrusaSlicer #12784: Brim being printed with wrong extruder on XL

fscheps, 2024-06-07:
### Description of the bug

I needed to print multiple objects, half of them in one color, and the other half in another color. 
But for some reason the brim of the objects on the left is being printed with the extruder of the right. Even if I have selected different extruders for different objects. Quite odd.
[DualPrint.3mf.zip](https://github.com/user-attachments/files/15712021/DualPrint.3mf.zip)
[image]

Adding the 3MF, a screenshot and a picture of the outcome, hope it helps to find a solution.

Thank you! 😉 
[image]

### Project file & How to reproduce

[DualPrint.3mf.zip](https://github.com/user-attachments/files/15712576/DualPrint.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+MacOS-arm64

### Operating system

14.5 (23F79)

### Printer model

Prusa XL dual toolhead

upgrayed13, 2024-11-05:
I also have noticed the same issue using Prusa Slicer Version 2.8.1+win64 with a Prusa XL with 5 print heads. In my case, I set the `Perimeter extruder`, `Infill extruder`, and `Solid infill extruder` to use extruder 2, while `Support material/raft/skirt extruder` and `Support material/raft interface extruder` were set to extruder 4.

I expected the brims to use extruder 4, but there's nothing I can do to change them to use extruder 2. I've tried enabling the brim globally, as well as for each individual object, but both yield the same results of using extruder 2.

Interestingly, the slicer does use extruder 4 for the supports, so it's only the brim that's effected. This further leads me to believe this is a bug.

My best guess for what is going on is that there is a bug where the slicer isn't actually using the `Support material/raft/skirt extruder` setting and is always leaving at the default (2) or setting it to 0 (i.e. use the "current" extruder), however that's just a guess.

upgrayed13, 2024-11-05:
If I'd have to guess, this issue is related to #12752.

wscullen, 2025-03-18:
I'm having an issue with multi-material on the prusa xl where the brim is using the same extruder as the support material, but I want the brim to be the same material as the main object, not the support material. I think a solution for that situation would be to allow a separate field for brim extruder, or have the brim default to the object material instead of the support (material) material. Because I think it's more common for the brim to be part of the object than to be part of the support.

aripperden, 2025-06-08:
> I'm having an issue with multi-material on the prusa xl where the brim is using the same extruder as the support material, but I want the brim to be the same material as the main object, not the [...]

I am experiencing the same behavior on a non-Prusa printer.

DanCrank, 2026-01-16:
Issue still exists in PrusaSlicer 2.9.4. I have an object printing in PETG with PLA support (using the soluble support setting) but also needs a brim. The brim prints with the PLA extruder regardless of other settings. (MK4 with MMU3)

closed 2026-09-29 by prusaslicer-bot[bot]