prusa3d/PrusaSlicer #12790: Small Area flow compensation

marcosalonso96, 2024-06-08:
**Is your feature request related to a problem? Please describe.**
A feature aimed at addressing the challenge of over-extrusion in small areas of solid infill, an issue previously not well understood. By implementing a strategy that adjusts extrusion flow inversely proportional to the length of the extrusion line, we ensure that shorter lines receive less flow. This approach is visually represented in the software, with color coding to clearly show the effect of the adjustments. This feature is designed to enhance print quality by providing a practical solution to the over-extrusion problem, making it easier for users to achieve consistent results across their prints.

**Describe the solution you'd like**
https://github.com/Alexander-T-Moss/Small-Area-Flow-Comp
https://github.com/SoftFever/OrcaSlicer/pull/3334

TheSlashEffect, 2024-06-21:
Before noticing this issue, I have ported this feature to PrusaSlicer in one of my forks.
Revision 904d2b5bf76afb9c3a3c631b37fdae42af40ff7c has the GUI implementation seen in OrcaSlicer.
Revision c94d238bdb95f1bf7add0399b1819bf9c242ad47 has an alternative implementation with a prettier GUI view, featuring one input field for each model value.

TheSlashEffect, 2024-06-21:
Fork can be found here:
https://github.com/TheSlashEffect/PrusaSlicer-SmallAreaInfillFlowCompensation

AndrewFarley, 2024-06-25:
@TheSlashEffect Any chance you can submit a PR?  ;)

TheSlashEffect, 2024-06-25:
@AndrewFarley hi, I'm not sure how common it is for user made features to be merged into main PrusaSlicer releases;
In addition, the feature is still under refinement based on user input (in the OrcaSlicer branch) so let us wait for a bit and then I'll gladly create one. 
Everyone is welcome to share their input on [this discussion](https://github.com/SoftFever/OrcaSlicer/pull/5838)

You can build a version on your own for now though

Schild0r, 2024-07-24:
IMO a somewhat clean solution in the interface would be if there is just a checkbox for this option and a button, that will open a window (like the button for the bed dimensions) with the model parameters + a graph for visualisation of flow over extrusion length.
Additionally the window could include model sugestions or a function input, that would calculate the model parameters based on an equation (after which they can still be adjusted manually).

TheSlashEffect, 2024-07-31:
Hi! Your first point is a good one, I have ported this to SuperSlicer and the owner of the project implemented exactly what you're talking about afterwards. You can check the [release](https://github.com/supermerill/SuperSlicer/releases/tag/2.5.60.0) out.

If those changes are compatible and I find the time I might port them here.

I am not sure what you mean in your second point though. Can you give some examples, maybe drawing it on pen and paper?

Schild0r, 2024-07-31:
Hey, this indeed comes really close to what I had imagined.
I noticed that it is probably not user friendly for most users to have the value pairs of the model calculated based on a formula (e.g. something like _y=1/-(2x+1)^4+1_ y: flow modifier; x: extrusion length; click a "calculate" button and it automatically populates the value pairs) so scrap that idea of mine.
[image]

The only things I would like additionally though would be 

- numeric input fields, that correspond to the handles on the graph (changing the handles in the graph, alters the values in the input fields and vice versa)
- add another "type": a bezeir fit across all points could result in a smoother transition than the quadratic interpolation between two points
- instead of using the cursor position inside the graph to display the x and y value in the bottom right corner, add something like a point, that "rides" on the graph at the x position of the cursor and display its values 

I created a quick power point mockup to show what I mean:
[image]

TheSlashEffect, 2024-08-04:
I agree with the idea about the input points, that's how it's implemented in my branch on PrusaSlicer atm. Bezier fit also seems nice, and the floating cursor thingy as well.

I took a look about porting the graph changes here, seems doable but would depend on the work on another feature we started there though, as well as some other obligations of mine. I'd like to see this ported to PrusaSlicer as well, so I'll update if I can get started on that.

AndrewFarley, 2026-09-05:
Functional PRs of implementing this are above, and would be welcome for this use case.  Please keep

oleksii-beep, 2026-10-01:
Hi @TheSlashEffect Do you have any kind of not-so-old version of fork? I'd like to try it, but if there is something recent I'd rather not merge on top on current PrusaSlicer myself

TheSlashEffect, 2026-10-02:
@oleksii-beep Hi there! I do not, I'm sorry. I only have a build environment for the 3.x stuff so I can't try it out at the moment.

You can try merging to the last build before 3.x, it seems it's doable...
https://github.com/prusa3d/PrusaSlicer/compare/version_2.9.6...TheSlashEffect:PrusaSlicer-SmallAreaInfillFlowCompensation:master

If not, the existing code I touched for these changes is not that much so it should be quite doable. The logic of the feature is isolated in separate classes.

TheSlashEffect, 2026-10-02:
@AndrewFarley thank you for keeping the request alive