prusa3d/PrusaSlicer #12791: Wipe while retracting disabled in Filament Override settings but slicer generates wipes

Martin-Danger, 2024-06-08:
### Description of the bug

Slicer is generating wipes in gcode. 
Printer settings has wipe while retracting enabled. 
Filament override settings has wipe while retracting disabled. 
[image]
[image]
[image]

### Project file & How to reproduce

Check printer settings enable wipe while retracting.
Check filament override settings disable wipe.
Slice model. 
Check gcode for wipes.
[Wipe disabled but still generates.3mf.zip](https://github.com/user-attachments/files/15746494/Wipe.disabled.but.still.generates.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Mac OS 10.15.7

### Printer model

Prusa i3 mk3.5

rtyr, 2024-06-08:
> Filament override settings has wipe while retracting disabled.

That is not the case based on your screenshot.

This is how it should look like (wipe override enabled, wipe setting disabled).

[image]

closed 2024-06-08 by rtyr

Martin-Danger, 2024-06-12:
Ahh, ok, thank you :-)