prusa3d/PrusaSlicer #12794: A single part print speed is slower compare to printing multiple parts together

Oron2000, 2024-06-08:
### Description of the bug

The print speed per part is faster when printing several parts simultaneously. How can I print a single part just as quickly (30mm/s instead of 10mm/s)?

This happened when I used a relatively small part (a 25mm diameter cylinder). Adding instances or increasing its size enables a faster print speed, up to a maximum print speed limitation of 30mm/s.

### Project file & How to reproduce

[https://youtu.be/8cKqeYVsAyI](url)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

win 11

### Printer model

ender 3 s1

Oron2000, 2024-06-08:
https://www.youtube.com/watch?v=8cKqeYVsAyI

Oron2000, 2024-06-08:
[image]
[image]

Oron2000, 2024-06-08:
Looks like it is a matter of the total bounding box of the printed parts:
Near parts are printed slower than far parts.

[image]
[image]

plampix, 2024-06-08:
I think you're overthinking this. It's probably related to your minimal layer time, causing the printer to print slower if this time would be exceeded.

Oron2000, 2024-06-08:
[image]

Many thanks, learned something new :)

closed 2024-06-08 by Oron2000