prusa3d/PrusaSlicer #12797: Select surface open, infill visible

jzagaja, 2024-06-09:
User select surfaces that want to be open and infill visible. This is usefull for acostics needs or pouring eg. resin to strenghten the infill.

jzagaja, 2024-06-10:
Current modifier removes perimeter but also add elswhere:

[image]

jzagaja, 2024-06-10:
Example, blue surfaces I want infill to be visible:

[image]

u89djt, 2024-06-10:
(fellow user interested in printing speaker enclosures) as I understand it, your only option would be to build two models that coincide: one to provide the integument for the infill volume you want access to, and one for that infill. Place those occupying the same space on the build plate. The slicer creates separate parts for each combination of infill and perimeters settings because switching between the two essentially demands an interface layer, which is provided by perimeters that are known to print well. So you need to provide a complete description of a shell around infill if you're to have any change. But - and it's a significant but - you won't succeed in pouring through the infill. If you need to be convinced, print a box with zero top layers and try filling that. Better to leave voids and fill those, e.g. https://www.youtube.com/watch?v=XEspOD1NHr0

jzagaja, 2024-06-10:
Solidworks wont create proper shell for perimeters. Sometimes it can but mostly refuses. It's tedious job. Better implement in slicer by selecting surface.

u89djt, 2024-06-10:
Here's roughly what your example would look like with infill that might allow filling with a curing/drying fluid:
[expose infill v1.zip](https://github.com/user-attachments/files/15769847/expose.infill.v1.zip)
[image]
[image]
I should imagine for a contributor to consider developing code to do this, they'd want to see that there's a popular application for it, and that the extrusion outcomes would be plausibly stable. 
If you can put the effort into building a CAD model that demonstrates what you want the requested slicer functionality to achieve, there might be a chance. It's possible you might find a willing collaborator if the CAD task requires different skills than yours currently. If you have a completed enclosure design, you could post a STEP or Solidworks here. I'm likely to take a look, for example. Maybe include bodies labelled "perimeter deletion".

jzagaja, 2024-06-10:
no oroblem

closed 2024-06-10 by jzagaja

u89djt, 2024-06-10:
Ah, that's not going to get the engagement you'd be looking for. The model will have to be a finished design for the enclosure in the form of shells, infill volumes and your intended shapes to impose absence of perimeters. When you're asking for a feature, it has to be absolutely explicit about what you want the result to be. Clearly I understand what you're getting at in general, but I think it's not a practical approach, and it wouldn't be used, so I wouldn't spend time developing code for it. You need to motivate someone to take an interest, which will involve a fully detailed presentation of what you're hoping will be the outcome. In this case, you can do it by hand, so you'd CAD the model in the way I described, import it into the slicer and test it, and then provide the CAD and 3mf files and screenshots of the important areas so the coding volunteer or product developer might believe it's worth trying, and doesn't have to make assumptions and do unnecessary work. If it wasn't possible to achieve with current software, you'd have to put together diagrams that mean the right things to the right people.

u89djt, 2024-06-10:
I just saw that you referred to thickening. There's a shell function in Solidworks: https://www.youtube.com/watch?v=w6RdBJiZPkg

jzagaja, 2024-06-10:
no - surface offset plus thicken. Just make modifier able to remove perimeter without adding one somewhere else. Easy? Then it will work as intended.

u89djt, 2024-06-10:
Good luck.

jzagaja, 2024-06-10:
Proably only Simplify3D can do this easily because uses processes. But 199$ for having gyroid...

jzagaja, 2024-06-12:
Modifier doe not differ much here from negative volume. In my opinion adding option remove perimeter should be added. Topic is not new:

https://community.ultimaker.com/topic/40744-removing-walls-aka-perimeters-in-specific-places-printing-speaker-enclosures/

u89djt, 2024-06-12:
Just looking at your previous submission https://github.com/prusa3d/PrusaSlicer/issues/10526
You can follow the advice in the forum thread you've linked to in prusaslicer.
[avoid.zip](https://github.com/user-attachments/files/15804377/avoid.zip)
[image]

jzagaja, 2024-06-12:
I can't get perimeter removed by using modifier - checked in PS, SS, Bamboo, Orca.

u89djt, 2024-06-12:
Yes, you'll have to dremel a face out as described by the forum thread.

jzagaja, 2024-06-12:
It's not an option and it comes from Cura.

u89djt, 2024-06-12:
I'm not sure what you mean. In the examples you've shown, and in the forum thread you linked to, cura has introduced walls around the modification.
This example of yours where there's exposed infill actually has perimeter walls cradling the volume of exposed infill.
[image]
That exposed modifier infill doesn't communicate with the infill in the rest of the object.

jzagaja, 2024-06-12:
I cannot replicate above.

u89djt, 2024-06-12:
It's your example from here: https://github.com/prusa3d/PrusaSlicer/issues/10526

jzagaja, 2024-06-12:
yes but I can't replicate in Bamboo Studio.

u89djt, 2024-06-12:
What do you mean by "comes from Cura"?

jzagaja, 2024-06-12:
In the ultimaker link they use Remove Mesh Intersection but I don't see any benefits.

u89djt, 2024-06-12:
If you don't remove the intersection between the mesh bodies, the region where they overlap will be printed twice on top of itself.
What do you mean by "comes from Cura?"

jzagaja, 2024-06-12:
You wrote "Yes, you'll have to dremel a face out as described by the forum thread."
In that ultimaker cura thread linked he uses Dremel so I wrote the idea "comes from Cura"

u89djt, 2024-06-12:
I don't think prusaslicer has any boolean operators such as intersection yet. You can do it in Solidworks.

jzagaja, 2024-06-12:
So far I could not do this in Solidworks that's why asking PS dev if they can turn modifier to remove perimeter as intended not adding elswhere.

u89djt, 2024-06-12:
OK, so it seems clear that the slicer developers are not going to give you that. None of the existing slicers do it, so you need to learn some more about Solidworks. You said you've been offsetting faces and thickening. Instead, try making a solid whole body, then using the shell function to turn it into a box with walls of defined thicknesses.

jzagaja, 2024-06-12:
How many request you need to programme it? Looking at other slicers is not the best idea.

u89djt, 2024-06-12:
(as I said in my initial response: fellow user interested in printing speaker enclosures. I'm not affiliated with Prusa) Requests aren't working. That means there aren't any developers who think it's a priority over everything else they're working on. So you need to provide evidence that will persuade someone that they should put in the development effort. If another slicer does do it, complete an example in that slicer. If there aren't any other slicers that do what you want, find a way of demonstrating that what you want is practical by completing an example, and gather opintion from other speaker enclosure builders. You can learn how to make a shell of your enclosure in Solidworks. It is definitely possible. It's a run-of-the-mill operation. My opinion - which is not necessarily more valid than yours - is that controlling the print by tuning the geometry of the walls and infill or support members in CAD will give the best outcome.

jzagaja, 2024-06-13:
another similar request: https://forum.prusa3d.com/forum/prusaslicer/adding-perimeter-modifier-creates-new-internal-perimeters/

u89djt, 2024-06-13:
Indeed. The more unfilled requests of this nature you can find, the more certain you can be that with the available information, no developer has been able to decide that implementing the request is worthwhile or practical.
You will need to provide different, incontrovertible evidence that simply removing part of the wall will still enable stable extrusions in the general case. I will stick my neck out and say that that is definitely not the case. You may find cases where it does work, but that is not the point for a developer. They need to see where and how it fails, so they would have to spend a lot of time exploring. Although I think they would very quickly hit messes that they don't want typical users to see coming out of their slicer on the print bed.

jzagaja, 2024-06-18:
Fusion has volumetric lattice but I haven't yet installed or tried slice.