prusa3d/PrusaSlicer #12799: Support vertical overhangs (arc overhangs)

jormundgand, 2024-06-10:
Vertical overhangs, overhangs that extend 90 degrees from the model, is possible by using a technique known as arc overhangs. The principle is to print a small overhang, an arc, and then extending it, by adding a new line of material on the edge of the already existing overhang, gradually building up the full shape line by line.

Work has already been completed that adds this feature to prusa slicer, in an outdated version of prusa slicer and as a python post-processing script.

For this to be really useful, a few kinks has to resolved regarding warping, something that already has seen some attention from the implementations already made.

The following results seem to be the bare minimum required to have a fully functional feature:
1. Be able to generate overhangs that extend 90 degrees from the model.
2. Be able to generate a space filling pattern like a Hilbert's curve as the first layer on a vertical overhang.
3. Be able to automatically generate support where such supports would reduce the risk of warping significantly.
4. Set parameters for vertical overhangs appropriate for those overhangs.

#4 is particularly difficult, as it would at least depend on the machine's cooling capabilities and the material used.

It is _technically_ enough to just be able to generate the overhangs without automatic support of alternative patters for the first layers, as long as people can add supports manually, but without #2 or #3, the feature is really not that useful.

It might be worth figuring out if it's possible to use something other than arcs and when that would make sense.

[One article on the topic](https://www.cnckitchen.com/blog/arc-overhangs-a-new-way-of-printing-without-supports)

neophyl, 2024-06-10:
This was already requested here https://github.com/prusa3d/PrusaSlicer/issues/9243  please search for existing issues before opening new ones.

Jan-Soustruznik, 2024-06-17:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9243

closed 2024-06-17 by Jan-Soustruznik

jormundgand, 2024-06-19:
> This was already requested here #9243 please search for existing issues before opening new ones.

I really shouldn't reply to this, but I will anyway. I did search for this and did quite a lot of searching too. I don't have time to look over 3900 issues and github search sucks. Please just close a ticket as "duplicate" instead when you see duplicate.

neophyl, 2024-06-19:
I can’t close it as I didn’t open it. Just another user like yourself. 
Also it took about 1 minute to find. If you think GitHub search sucks you should try the Wordpress one used on the Prusa forum lol. You would be praising the GitHub one.