prusa3d/PrusaSlicer #12801: Specific Infill Tool (XL/MMU)

easycheese, 2024-06-10:
We all want large prints on the XL to look good, so we use expensive filament. But there is no need for the filament on the inside to be expensive. 

We should be able to select a specific tool to be dedicated to infill (similar how we can do for supports). I could save 50% or more on larger prints by swapping out $60 filament for $20 filament on the infill.

Note that while this is applicable to the MMU, the time costs probably make it not worthwhile (unless that color was already a part of the object). The XL is where this can really shine.

u89djt, 2024-06-10:
(Intrigued fellow user) Fooling around to try to get this to happen, I've got the following:
[2 different infills.zip](https://github.com/user-attachments/files/15778178/2.different.infills.zip)
[image]
There are general print settings that you would imagine sorting this out,
[image]
but I've only got it working with added settings in the object list so far.

wafflecart, 2024-06-11:
Confirm it can be done and I’ve used it before with similar settings to u89djt although I keep the “solid infill extruder” the same as perimeter extruder, this is the top and bottom walls.

See it in action here:
https://x.com/wafflecart3d/status/1789946921588531640

easycheese, 2024-06-11:
> Confirm it can be done and I’ve used it before with similar settings to u89djt although I keep the “solid infill extruder” the same as perimeter extruder, this is the top and bottom walls.
> 
> See it in action here: https://x.com/wafflecart3d/status/1789946921588531640

Well now this is interesting. I can get it to work on a cube shape but not on my other file.

u89djt, 2024-06-11:
Yes, if your object is set to use default extruder, it'll use the default infill extruders set in the general settings. If you set a specific extruder for the object, that extruder will be used for all aspects of it including the infill, unless you add a setting to that object to override it. You have full control, but that means you have to exert that control unless you leave everything at defaults :)

easycheese, 2024-06-11:
Ok I figured out how to replicate. Seems like a bug. If you use the bucket fill to define color, the infill gets ignored. See the following. One cube was left as the default, the second cube was bucket filled to a different color. In my opinion, bucket fill should only effect the exterior, not the infill.

[image]

u89djt, 2024-06-11:
Maybe if that's all you apply to the object, in which case you might as well set the overall tool for the object. If it's bothering to bucket fill, then parts of the object are going to be painted differently, and starting different extrusions on an outer surface is pretty fraught. I encountered this when dividing up Emmett's heart and ended up with tips of teeth different colours, and that just didn't work in many cases. Having part of that extruder's exclusions plunge into the rest of the body is presumably the best general solution the slicer developers have come up with. Adding supports doesn't guarantee success either.

easycheese, 2024-06-11:
Yeah so Shift + Left Click solves the problem on the original file.

But it occurs to me that my original statement, while resolvable via this solution isn't perhaps the 'best' answer. Forcing the infill to use an extruder that may not have been used on that layer is not efficient. Really the best answer is a 'dont use this extruder for infill' option that could be selected on the extruders that are using expensive filament. This way, the auto select for infill could still choose the efficient infill tool (one already being used).

afreiberger, 2026-02-04:
Adding to the above ideas another use case to consider.  I have an object with two colors, one color is a color-shifting rainbow and I want it to be the infill so I move through more of the color shift.  When I painted my object the two colors and set the infill extruder to the rainbow extruder, it still does a half and half color infill.  So, in addition to the cost driver, it'd also be useful in a situation where you want to make it through more of one spool than another.

closed 2026-09-29 by prusaslicer-bot[bot]