prusa3d/PrusaSlicer #12808: When opening a project, nothing is shown...

FredM67, 2024-06-12:
### Description of the bug

When opening a project, sometimes, nothing is shown. The Plater stays empty.
After slicing, the 3d-model appears.

### Project file & How to reproduce

[SmartTemperatureTower_Stand_3DJake_ASA_Black.zip](https://github.com/user-attachments/files/15806020/SmartTemperatureTower_Stand_3DJake_ASA_Black.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 11 23H2 (OS Build 22635.3720)

### Printer model

Snapmaker Artisan

neophyl, 2024-06-12:
Before you open a project are you on the 3d Editor or the layers preview ?  

If you are on the layer preview then it will be empty until you slice as there is no preview generated of your newly opened project until you do so. 
Loading a project does not automatically switch you back to the 3d editor view.  

https://help.prusa3d.com/article/ui-overview_1766

FredM67, 2024-06-12:
I was on the Preview mode !
When I switch to 3d Editor, I see the model. When I switch back to Preview, slicing is automatically done and I see all the layers.
Then I open another project (the one attached in the Issue). Again, I don't see anything in Preview.
I switch to 3d Model, I see it.
Then switch back again to Preview, slicing is done, BUT I only see the first stl sliced !! And somehow, there's no way to get it complet anymore except closing the application !

closed 2026-09-29 by prusaslicer-bot[bot]