prusa3d/PrusaSlicer #12811: Allow for color change when printing sequential objects

mkatzpdp, 2024-06-13:
When printing multiple objects sequentially allow for a color change (either with MMU or without) between objects.

When printing single color objects sequentially where the next object is a different color than the current object, allow for a color change (with or without MMU).  This would include a small wipe tower when using the MMU.

Github won't let me attach a .3mf file.

To create this issue, just put two or more objects on the plater, set them to different colors and enable sequential printing.

neophyl, 2024-06-13:
Duplicate of already existing issue  https://github.com/prusa3d/PrusaSlicer/issues/7404 

For future reference to attach files you need to zip them up first.

closed 2024-06-13 by rtyr

mkatzpdp, 2024-06-13:
This issue is not the same as #7404.  Though it is similar.  All of the features here could be implemented for units without an MMU.  In that case, this issue should replace #7404.

Differences:
#1 - The issue here is to allow automatic color change via the MMU when sequential printing is requested.  There is no mention of the MMU in #7404.

#2 - Update the slicer to automatically recognize that there are multiple objects (each of a single color in the object) on the plater but each object is a different color.  This should bring up a prompt asking if sequential printing should be enabled.

#3 - This will require a small wipe tower to clean the extruder only between objects.

mkatzpdp, 2024-06-13:
Please reopen this issue.

rtyr, 2024-06-13:
https://github.com/prusa3d/PrusaSlicer/issues/1787