prusa3d/PrusaSlicer #12813: [BUG] Solid infill line in the air affects bridging

vovodroid, 2024-06-13:
### Description of the bug

For whatever reason solid infill line is added to the wall:

[image]

And bridge on the next layer fails to connect to the same wall:

[image]

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/user-attachments/files/15823324/Shape-Box.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Win10x64

### Printer model

Home made

u89djt, 2024-06-13:
It does it at both ends
[image]
And the bridging infill attaches to that
[image]

vovodroid, 2024-06-13:
> And the bridging infill attaches to that

On the picture yes. In reality not. Solid line in the air is not good for bridge attaching.

u89djt, 2024-06-13:
It seems to be working as intended for me:
[image]
[image]

vovodroid, 2024-06-13:
> It seems to be working as intended for me:

Well, it also depends on a lot of other options. Nevertheless extrude near 100% overhang line and then attach bridge to it is problematic approach.

u89djt, 2024-06-13:
That's printed with your options. The overhang line is absolutely fine. 
[image]

vovodroid, 2024-06-13:
> That's printed with your options.

Not really, I just created empty project with standard settings. Even it's fine with some settings, why do it and complicate things?

u89djt, 2024-06-13:
You want an explanation of why an anchor is printed for the bridge infill? OK. You'll be aware that all infill is started with an anchor at the wall, and that anchor extrusion overlaps the inner perimeter by 15% by default. That's what's happening here. Bridging infill is a higher volume extrusion run more slowly with plenty of cooling. The anchor provides more of a cradle against the wall for it to adhere to as it pulls across the gap. Mashing the briding extrusion into the wall proper instead of a dedicated lip first would make a mess or be more likely to fail to adhere.

vovodroid, 2024-06-13:
I have no problems with even long bridges connection to wall (two perimeters), but this case with solid line fails. Probably it should be configurable.
What is logic in the code to add such line in certain cases? How can I disable it?

u89djt, 2024-06-13:
Can you post an example 3mf that succeeds for you? 
I haven't looked at the code.
Consider drying your filament and calibrating its extrusion multiplier. If it's under-extruding, or sputtering while it extrudes, maybe that cause adhesion to fail.

u89djt, 2024-06-14:
Are you saying that the bridges that succeed for you are external, not internal?

vovodroid, 2024-06-14:
> Are you saying that the bridges that succeed for you are external, not internal?

Internal without solid line succeed as well. The only problem is regions with solid line added to the wall.

u89djt, 2024-06-15:
This means your printer or filament isn't working like mine. The brindging infill between lines works fine. Maybe your filament is under-thickness and so doesn't reach as far out to adhere to walls as it should? That would be fixed by calibrating the extrusion multiplier. Am I over-extruding? The filament I used is my favourite reliably calibrated filament and you can see that the anchor line printed well. Perhaps this kind of infill anchoring is a problem for lower quality filament? What are you using? Is your printer tuned up? Proper belt tensions and so forth.
Clearly identifying at least one specific example where it fails will be key to gaining traction. I don't believe I'm the most successful printer here: the one you posted is not going to fail for people in general. 
I had a look at what external bridging would look like in your object by hollowing out the base. You'll note that internal bridging infill is much thicker than external bridging now that it is not thick by default.
[hollowed.zip](https://github.com/user-attachments/files/15847864/hollowed.zip)
[image]
Maybe the thicker extrusions are pulling harder as they cool. That could pull the line off the wall if that line was under-extruded and therefore didn't mingle well enough.
How about this: show a failure in detail by printing an object and stopping the print when the bridging fails so that it can be seen in a photograph.

staal54a, 2025-05-03:
Found this and I agree with the original bug and the way it is still implemented, it causes issues and at best is a very poor implementation. I think something along the lines of https://github.com/prusa3d/PrusaSlicer/issues/12144 would be a much better option. 

The problem is that the anchor uses the solid infill type which in 99% of cases is printed over either a solid layer or a bridge layer. That means in a lot of profiles, speed can and is increased for solid infill. However, when the anchor is being laid down by the printer, it is basically equivalent to an overhang (I'd say it's worse than an overhang because for at least part of the anchor it is only attached to the line adjacent to it and when printing with 0% infill, there are no sparse infill lines supporting it). Normally for overhangs, we slow down to allow the plastic to cool and properly adhere to the previous layer and printing too fast can cause issues and curling.

So, in my case, what happens is I will have basically a perfect print up until I need some bridging internal to my part's geometry. Then, when it gets to the layer before the bridge, this anchor causes a lot of fast print moves without the proper support underneath to occur and pulls lots of filament strands upwards (the infill anchor being moved inside also has similar issues where it isn't supported very well depending on pattern and density so it doubles the potential for problems). Generally, it usually only causes some nozzle buildup for myself since I tend to print smaller parts and clean the nozzle religiously before prints. It does also tend to be more problematic with certain filament types. So I see it have more of an effect with PETG vs PLA for instance.

However, I would argue it still is problematic as nozzle buildup would eventually cause issues over time, such as on a longer, complex print. The bits of filament that get lifted up could also cool and serve as point the nozzle could hit to dislodge a print.

I think the minimum solution should be to change the line type of the anchor to be recognized as a bridge or overhang and be printed with a much slower speed to allow it to adhere to the adjacent wall. I think that would resolve a lot of the issues I see myself and I think others do as well.

I think a better solution is to re-evaluate the anchor solution and look at something like the feature request I linked above where you could allow users to specify how much over the current layer's perimeters a bridge could overlap (similar to infill/perimeters overlap). This would still make use of the same system of trying to adhere to an adjacent line but would also take advantage of the typically slower bridging speeds to allow it to adhere to that adjacent line/wall.

(didn't proofread so probably some grammar errors above, will edit if I see them later on).

closed 2026-09-29 by prusaslicer-bot[bot]