prusa3d/PrusaSlicer #12815: Suggestion: Automatically increase extruder clearance height if necessary to complete objects individually

samyk, 2024-06-14:
**Is your feature request related to a problem? Please describe.**
Hi! I was trying to print several items sequentially using the `Complete individual objects` option and confusingly got a "Some objects are too tall and cannot be printed without extruder collisions" error. Even if I placed the two objects in a row where one could be completed first, then the next one entirely behind it with a gap, the message persisted, and I tried swapping their positions to no avail, though I think technically it would be fine (at least in that specific scenario) as long as the front-most print is done first.

**Describe the solution you'd like**
It seems that even if that's not possible, just increasing the `Extruder clearance height` resolves the issue. Instead of outputting the error, it would be nice if the software simply increased the extruder clearance height to the minimum height necessary to perform the print as it was not clear at all that this was possible. I understand this takes (much appreciated!) effort to implement, so even in the interim a more verbose message explaining that this height can be adjusted to potentially resolve the issue would be nice so it's clear what to do next rather than bailing on the print or going web sleuthing for potential solutions.

**Describe how it would work**
More descriptive error as an interim solution, extrusion height modification for just the specific print if automated.

**Describe alternatives you've considered**
Alternative is better error messaging.

Thanks!

u89djt, 2024-06-14:
Go ahead and post the 3mf file and someone (possibly a user like me or neophyl) will run through what's happening. There are several similar queries like yours on github. I don't want to assume you've missed what's actually happening like everyone else (including me early on!) so I'll look at your project file. It's very easy to place things on the bed such that simply increasing the extruder clearance height parameter will have the cross members crashing into objects. That parameter doesn't tell the extruder how high to move between objects, it describes how far the cross members are above the nozzle tip.
Here's (one of?) Prusa's article https://help.prusa3d.com/article/sequential-printing_124589 

neophyl et al, I have a plan - build an object to move around the build plate that demonstrates and allows testing of the volume of the extruder and swept plane of the cross members. The prusa article about it provides the necessary information, but it doesn't teach it per se. I'd be in favour of writing one. It'll be a more involved process than answering specific queries on here so I won't promise when.

samyk, 2024-06-14:
[2parts-tootall.3mf.zip](https://github.com/user-attachments/files/15841279/2parts-tootall.3mf.zip)
[image]
Example attached - I found if I adjust the extruder clearance height from default of 14mm to 20mm, the error goes away and I can slice

u89djt, 2024-06-15:
Good stuff. In that example, when the printer is printing the second object, the X axis carriage won't hit the first object. Let's see how our impressions compare about what would happen in this case:
[image]
Because the distance between the tip of the nozzle and the long horizontal underside of the X axis carriage is 14mm, and the objects are 19mm high, when attempting to print the second object, the X carriage would sit on the first object and stop the tool being lowered to the bed or any of the first 5mm of layers.
Changing the clearance parameter to 20mm would stop the slicer warning you, but it wouldn't change the physical impossibility of performing the print. The slicer does not currently include logic to recognize the difference between your successful case and an impossible case. To work out whether a given configuration of objects can be printed sequentially with something more sophisticated that the radius of the peeking cylinder and horizontal plane of the underside of the X carriage would involve simulating the whole printing process in terms of motion of the appropriately modelled volumes of the tool and carriage. Eminently feasible, but a lot of development work, and a much longer slicing period, or maybe a sanity check phase. Certainly people more in the know than me will be interested in having this function. There is discussion elsewhere about completing vertical ranges of individual objects rather than single layers then moving on during a general print. For with the example I posted here, you could print say the first 13mm of object one, the whole of object 2, then the remainder of object 1.
For now, you need to lie to the printer about how the tool is mounted and apply your own knowledge about what's going to happen.

samyk, 2024-06-15:
I see, thanks for the explanation - I had to sort of sit at the printer to get a geometrical understanding of what was happening and under what conditions it could succeed or not and the failure vs success makes sense now.

Here's an interesting print that's working right now and is on the 4th item (set a 130mm clearance height) though I'm a little confused how the front two items printed since it appears they're about 30mm vertically (x-axis) so shouldn't the carriage have run into one of the front ones? I've attached the 3mf of just the front 2, I moved stuff around and then tried to Redo via cmd+shift+z (which is Redo in most apps on macOS) but realized it was undoing and broke the history at some point, but anyway the 2 front pieces are there which are the confusing ones to me.

I agree it would be cool if the slicer (and Arrange function for that matter) could simulate the carriage+nozzle+prints to find if there will ever be interference, then it could automatically increase the clearance height while testing, but I understand that's additional work (and I have no concept of the complexity of that).

If I'm understanding the challenges, Arrange perhaps could do a few things to improve likelihood of success:
- rotate objects to maximize horizontal space while minimizing vertical (x-axis, not height) space on a per-object basis
- placing tallest items closest to the front
- place ample space to prevent carriage to object collisions
- place items of similar vertical distance (x-axis) along the same vertical line - for example I imagine the print above could have all been done next to each other across a line, at least if they're vertically short enough that there won't be carriage running into it
- increase extruder clearance height to `tallest object height + 1mm`
Though I'm sure this is a lot of work and easier said than done!

[image]
[2parts-front.3mf.zip](https://github.com/user-attachments/files/15855205/2parts-front.3mf.zip)
[image]
[image]

u89djt, 2024-06-15:
Great, will dig around tomorrow! In the meantime, was the one on the left printed first? I'd have to guess that the y range of the object is less than the y offset of the nozzle from the x carriage, so you'd have the object set nearer the edge than the first one, and the nozzle far enough y-ward that the x carriage doesn't meet the previous object. I presume there's a finite risk of my deleting this in embarrassment tomorrow :)

samyk, 2024-06-15:
Haha, well from the Prusa image I believe each square is 10x10mm so even if those two are off by 10mm from each other, then there'd be 23mm (the objects are 33x33x125.6mm) where there's overlap, and 23mm - 14mm(extruder-to-carriage) = 9mm of interference I'd expect, but it clearly printed so maybe I should take some measurements to try to figure out what's going on as I'm definitely confused!

samyk, 2024-06-16:
I just loosely measured from the Prusa XL nozzle tip to the edge of the x-carriage (closest to the nozzle) and it was around ~25mm which now makes sense why the print worked -- order in the universe restored. Your explanation really helped me understand what was going on, thanks!

u89djt, 2024-06-16:
Hey, fantastic! And good on you for keeping up the chase. This is how it's supposed to go. Should I invite you to comment on a draft writeup to offer to prusa for their article on sequential printing when I get to it?

samyk, 2024-07-05:
Sure, if I have time I'll be happy to review.

u89djt, 2024-07-05:
Great, I'll pick that back up.

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Your feedback is welcome. Closing.

closed 2025-02-18 by lukasmatena