prusa3d/PrusaSlicer #12823: Layer times are inaccessible from new slider (2.8.0)

dwineman, 2024-06-15:
### Description of the bug

In 2.7.4 and earlier, it was possible to hover over the gcode preview layer slider and see the elapsed time at which that layer would start printing:

<img width="151" alt="Screenshot 2024-06-14 at 5 10 10 PM" 
src="https://github.com/prusa3d/PrusaSlicer/assets/372994/1df4f6cf-73dd-4d7e-a2e7-91d745de23c5">

This information is no longer present in the redesigned sliders in 2.8.0 alpha:

[image]

There's even more room to display this info clearly now, so it's hard to understand why it was removed.

### Project file & How to reproduce

This is a general UI issue that is independent of the model or settings used. Reproduce by slicing any model at all and inspecting the slider. But here's a project file to satisfy the requirement. 
[sample.3mf.zip](https://github.com/user-attachments/files/15844977/sample.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5+MacOS-x64

### Operating system

MacOS 14.5

### Printer model

Prusa MK4

stevekenney318, 2024-06-15:
@lukasmatena -  In my opinion, this is the worst UI update of Prusa Slicer that I have ever seen. I do not like any of it. It looks like it was designed by and for children. From what I have seen,. I will not be moving forward to 2.8

dwineman, 2024-06-15:
I wouldn't go that far, but it's not really a matter of taste. I just don't want to lose functionality.

neophyl, 2024-06-15:
Looks like there's new settings options under the slider.  
[image]

If you click on that cog then there's an option to show the layer time.  I think that should restore your lost functionality.
[image]

stevekenney318, 2024-06-15:
> Looks like there's new settings options under the slider. 

Yes, that will bring back the functionality of the time. Actually that is a setting that was previously there before.

I personally don't like the look and feel of the new sliders... I prefer the look of the old design and appearance. I do not like it floating in space. I like it before in a defined darker space.

dwineman, 2024-06-15:
@neophyl Yes, thank you, that solves it. I hadn't noticed those options until now.

closed 2024-06-15 by dwineman

WhiteDragonD, 2024-06-16:
I just wrote up a bug for this too because I use those lines to 'count' how many top layers the second color will be.  Now I have to hold the slider button and move it and try to count instead of just looking at how many lines there are.  It's annoying.

stevekenney318, 2024-06-16:
> I just wrote up a bug for this too because I use those lines to 'count' how many top layers the second color will be.  Now I have to hold the slider button and move it and try to count instead of just looking at how many lines there are.  It's annoying.  

I agree. The interface was one of the best things about the software. I don't understand why they changed it.

cntbill, 2024-06-16:
> but it's not really a matter of taste. **I just don't want to lose functionality**.

kubispe1, 2024-06-19:
Most probably we will enhance it in next release: SPE-2353
thanks for your opinion