prusa3d/PrusaSlicer #12825: is it supported to install to the system multiple versions of prusaslicer?

fordfrog, 2024-06-15:
i wonder if there is a switch in the build system for installing prusaslicer to a different than the default path and versioned (desktop file etc) so that multiple versions can be installed on linux system side-by-side. i'm asking from the point of view of a gentoo developer, as i'd like to have packaged and installed both the latest alpha and the current latest stable side-by-side. we currently install either one or the other.

fordfrog, 2024-06-15:
i've read about the profiles being located in a different place. my question was rather about the binaries and shared stuff.

currently the layout on gentoo linux is this:
```
/usr
/usr/bin
/usr/bin/OCCTWrapper.so
/usr/bin/prusa-gcodeviewer -> prusa-slicer
/usr/bin/prusa-slicer
/usr/lib
/usr/lib/udev
/usr/lib/udev/rules.d
/usr/lib/udev/rules.d/90-3dconnexion.rules
/usr/share
/usr/share/PrusaSlicer
...
```
i was not able to install the alpha yet as we have an issue with finding wxwidgets:
```
CMake Error at cmake/modules/FindPackageHandleStandardArgs_SLIC3R.cmake:143 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES) (Required is at
  least version "3.2")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindwxWidgets.cmake:990 (find_package_handle_standard_args)
  src/CMakeLists.txt:52 (find_package)
```
this issue is not present in 2.7.x. anyway, because of this issue i was not able to check the installation paths and binary names. but in case alpha installs files to the same location and with the same names, we could not have multiple versions installed side-by-side.

SachCZ, 2024-06-17:
Hello @fordfrog, I am not entirely sure what are you trying to achieve. To try to help you, there are several facts that might hopefully useful to you:
- Prusa slic3r is relatively standard in its cmake use, so you should be able to use the `CMAKE_INSTALL_PREFIX`option to install it to wherever you like.
- This **might be obvious** but just to be sure: Prusa slic3r **does not have to be installed to be used**. You can use either the compressed files or the appimage downloaded from our github release page directly. Just download them and use them directly (`cd PrusaSlicer/bin && ./prusa-slicer` or `.../PrusaSlicer/bin/prusa-slicer`). The same thing is obviously true if you compile PrusaSlicer yourself. As developers we are usually juggling more than a few versions of PrusaSlicer.
  - you can add the bin folder to your path
  - you can trigger desktop integration from PrusaSlicer
- When compiling PrusaSlicer yourself it is preferred to also compile the dependencies yourself. There is a manual [here](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md).
- With the compiled dependencies make sure to specify the `CMAKE_PREFIX_PATH` when compiling PrusaSlicer.
- Make sure your dependencies compiled successfully. Some of the steps might have failed and this then results in your error. 
- In the new version there is a transitive dependency that cannot really be practically compiled as part of the slicer dependencies `libwebkit2gtk` - make sure to install it on your system.

I hope some of these points help you.

fordfrog, 2024-06-18:
@SachCZ thank you for your info.

in gentoo we use ebuilds to install packages. currently we have ebuild prusaslicer-2.7.4 and prusaslicer-9999 (which compiles and installs prusaslicer from the master branch, which is currently alpha5 afaik).

my question was, whether there is a switch that would rather cause prusaslicer being installed versioned. like for scribus, if we specify version during the build, we can safely install several versions of scribus on the user system.

the question was, whether it is possible to do the same with prusaslicer as we can do with scribus. that would imo mean:
- versioned files in /usr/bin
- versioned directory /usr/share/PrusaSlicer
- versioned files in /usr/share/applications and /usr/share/icons
- maybe some other stuff

so my question was whether prusaslicer build system supports such configuration (like specifying a version suffix) and i just overlooked that or it is not supported.

SachCZ, 2024-06-18:
Hello, I see what you are after. I am afraid it is not supported.

fordfrog, 2024-06-19:
> Hello, I see what you are after. I am afraid it is not supported.

oki, thanks for assurance. i'm closing this as you answered my question.