prusa3d/PrusaSlicer #12827: Numbers are missing from the slider and the info when hovering.

WhiteDragonD, 2024-06-16:
### Description of the bug

On version PrusaSlicer 2.8.0 Alpha 5, when I'm slicing an object with raised text or a layer/nozzle change, I used the slider to easily count how many layers will be used when adding text.  With the new slider, those numbers are gone in addition to showing the times when hovering.  

Please add those numbers back, the 2 mm of 'saved space' really isn't that helpful.  

It's a PAIN in the butt to hold the new button and drag it up layer by layer to count.

[image]

### Project file & How to reproduce

Add part, box, add 1mm of raised text.
Slice... notice the numbers and - are missing from the slider.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.8.0 Alpha 5

### Operating system

Windows 10

### Printer model

XL, 2.5s

neophyl, 2024-06-16:
Please see https://github.com/prusa3d/PrusaSlicer/issues/12823

kubispe1, 2024-06-19:
Most probably we will enhance it in next release: SPE-2353
thanks for your opinion

lukasmatena, 2024-06-21:
The estimated times overlay is enabled by default in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1).
The settings are now remembered between restarts.

Is the issue resolved?

WhiteDragonD, 2024-06-21:
The ruler marks are now there.. BUT... when you change extruders, the color isn't staying at the same level.   It's moving up and down w/ the slider.  

The m600 color is there though.

WhiteDragonD, 2024-06-21:
and the Ruler says: Show Backgroung  instead of 'Background'

kubispe1, 2024-07-11:
The original requirement has been implemented. At the moment, we don't plan any new improvements, only possibly some bug fixes. I am closing the issue. Thanks.

closed 2024-07-11 by kubispe1