prusa3d/PrusaSlicer #12833: Saving Prusa Connect passwort failed

CaveRenderPro, 2024-06-16:
### Description of the bug

Sign in to Prusa Connect works, but the passwort could not saved. 
[image]

If I start Prusa Slicer as Admin sign in to Prusa Connect does not work anymore

### Project file & How to reproduce

Saving Prusa Connect passwort

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Windows 11

### Printer model

MK4

kubispe1, 2024-06-19:
This could potentially be an issue with PrusaSlicer.
- When I run 2.8.0-alpha5 as administrator and sign in to my PrusaAccount via Google account in the browser, the credentials are not being passed successfully. 
- However, I did not receive any error message.

CaveRenderPro, 2024-12-20:
The issue is fixed with PrusaSlicer Version 2.9.0.
Thanks a lot.

araghon007, 2025-02-03:
Same issue at 2.9.0, not running as administrator