prusa3d/PrusaSlicer #12834: PrusaSlicer 2.8.0. Alfa5 MK4 + MMU3 Prusa Connect wronf type

Janek250, 2024-06-16:
### Description of the bug

in prusa connect have MK4 (I have MK4+MMU3), Slicer profile MK4 with MMU3, when click "send to connect" next dialog window show me "No compatibile printer" 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0. alfa 5

### Operating system

windows 11 home

### Printer model

MK4 MMU3

PatB42, 2024-06-16:
Hi 

Ok just install it, nicer UI , connect button ok but where is the G button to send to prusa link ?? my customer don't have connect and they must have an acc and too complicate for them that why link is very useful ...but as you are connected to your acc with the slicer then only connect appears !! it's alpha 5 ;-)

kubispe1, 2024-06-19:
It appears this issue is currently specific to the MK4 model with MMU3. We'll investigate it further in collaboration with the Connect team.
- Now, when you select a printer in the Connect tab and choose "Set as current," it will at least display the printer as your current printer when sending to Connect.
[image]

kubispe1, 2024-07-04:
AFAIK Connect fixed this bug and now it should work. Can you please confirm @Janek250

closed 2026-09-29 by prusaslicer-bot[bot]