prusa3d/PrusaSlicer #12835: Missing libwebkit2gtk-4.0.so.37

Pitel, 2024-06-16:
### Description of the bug

Can'r run 2.8.0-alpha 5 on Ubuntu 24.04.

### Project file & How to reproduce

```
/tmp/.mount_PrusaSxTFXZN/usr/bin/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```

And yes, I have `libwebkit2gtk` installed with `apt`.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Ubuntu 24.04

### Printer model

MINI

neophyl, 2024-06-16:
The release notes mention that it wont run on Ubuntu yet.  
"An important note for Linux users: The binaries will not run on newest Ubuntu 24.04 (and possibly other latest distributions). This problem will be solved shortly in the next alpha/beta."

closed 2024-06-16 by Pitel

lukasmatena, 2024-06-28:
@abatkin 

Thanks for the input. I have answered in #12931 to have the discussion in one place.