prusa3d/PrusaSlicer #12836: Ruler Mode missing scales on sliders...

cntbill, 2024-06-16:
**Is your feature request related to a problem? Please describe.**
Missing ruler mode with preview.... 

**Describe the solution you'd like**
Please make it an option maybe

**Describe how it would work**
Maybe just put option in settings gear on bottom 

**Describe alternatives you've considered**
Going back to PrusaSlicer 2.7.4

**Additional context**
I use the scales to set various color options and view times, having to move slider up and down to see them is a PiTA

dunkmann00, 2024-06-18:
I also noticed this problem with the new 2.8.0-alpha5 on mac. I'm hoping this is just a mistake? I definitely like having the ruler visible.

kubispe1, 2024-06-19:
Most probably we will enhance it in next release: SPE-2353
thanks for your opinion

lukasmatena, 2024-06-21:
Thanks for the feedback. Ruler mode is now available in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Closing.

closed 2024-06-21 by lukasmatena