prusa3d/PrusaSlicer #12837: 2.8.0-alpha5 add color change button VERY hard to click now

nathan118118, 2024-06-16:
### Description of the bug

Trying out 2.8.0-alpha5. The button to add a color change is not working well. In the past it was easy to mouse scroll wheel to the right height, click the plus, mouse scroll wheel, click another height.

But with the 2.8.0-alpha5, clicking on the plus sign now routinely does NOT add a color change, but instead changes the layer height. It's like the bounding box for clicking the plus sign is _really_ small...so if you don't have your cursor right in the very center, it thinks you want to jump to a specific layer height, when really you're trying to click the already very small add color change plus sign.

Never had this problem with the old style where the plus was off to the side, even though clicking above or below the old plus sign would select a specific layer height.

### Project file & How to reproduce

Screenshot included of old and new color change plus sign.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Windows 11

### Printer model

Prusa MK3s+

mix579, 2024-06-17:
Agreed. This is a step backwards in functionality. I'm on macOS.

lukasmatena, 2024-06-21:
This should be fixed in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Closing.

closed 2024-06-21 by lukasmatena

nathan118118, 2024-06-21:
Thanks Lukas! Appreciate you!