prusa3d/PrusaSlicer #12838: 2.8.0-alpha5 on Debian Linux: menu is gone, useful buttons are tiny, other buttons are gigantic

paulcobbaut, 2024-06-17:
### Description of the bug

As you can see in the screenshot with 2.8.0-alpha5 on the left and 2.7.4 on the right, the top menu has disappeared!
Second thing is that the most used icons on the top and the left side are now tiny compared to 2.7.4. (Also this grey on grey with transparent grey behind it is not the most readable colour scheme.)
Third thing is that the less used buttons for "print settings" etc are gigantic with a lot of wasted white space around them.

Please bring back the normal menu, please decrease the size of the "Plater" and adjacent buttons (or better make those a part of a normal menu), and please restore the original size of the commonly used icons on the top and the left.

[image]

### Project file & How to reproduce

Open the Appimage.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Debian

### Printer model

MK3S+ and MK4

MnlSx, 2024-06-17:
I totally agree on that. Please don't remove the system menu and hide it behind a button. It might look "more modern" but requires an extra click for accessing functions I use for allmost every print. 
Trade usabillity for style is not a wise choice in my opinion. 

I refer to the Windows version, but seems to be the same on Linux.

plampix, 2024-06-18:
Looks like the icons size setting in the GUI tab is ignored.

apopticon, 2024-06-28:
Another vote to restore the top menubar. Why add another click for everything? It's an annoying &  poor UI decision.  I don't know why anyone ever thought that it was a good idea. It's not like we need a UI designed for a smartphone.

Shinomorisama, 2024-06-28:
Also throwing in my support for this.  I use the menus with every print and the new system requires more clicks and more precise mouse movements making it more difficult to do tasks than before.   I don't have much opinion on the cosmetics but the menu change is a clear decrease in function.

SprocketTools, 2024-07-02:
IMO the old menu was better, having the big buttons is a waste of space for what they are.

CobraA1, 2024-07-04:
Agree. It's an extra click to do things. It's not even saving any screen space, since that's offset by the increased size (and the old way wasn't really wasting much space to begin with).

. . . and since the menu items are packed together closely when you open the menu, it's not even touch screen friendly.

Grasping at straws trying to find how this is an improvement - coming up empty. It's not an improvement.

MR2C280, 2024-07-07:
Normally I just automatically install new versions from Prusa because they've always been so reliable.  This is the first time they've really cocked it up.  Now I'm going to have to faff around rolling back to 2.7 until they get this fixed.

ColsonThePCMechanic, 2024-07-07:
Relevant to the conversation: https://www.reddit.com/r/prusa3d/comments/1dwpxza/comment/lbwg5vy/

lukasmatena, 2024-09-13:
I just closed #12943 (more in [my post](https://github.com/prusa3d/PrusaSlicer/issues/12943#issuecomment-2348361789) in that thread).
Closing this one as a duplicate. The part with toolbar icon size is tracked in #13000.