prusa3d/PrusaSlicer #12839: Organic support in part expands too much in z-axis.

MartenKL, 2024-06-17:
### Description of the bug

When slicing a part that has support in the part, the organic support pushes the part up too much causing a warp-like problem.
[Slicerissue_OrganicZExpansion.zip](https://github.com/user-attachments/files/15864328/Slicerissue_OrganicZExpansion.zip)

### Project file & How to reproduce

Print with Organic support and the midsection of the legs have expansion in z-axis, changing to Snug eliminates the issue.  Have eliminated all other possible solves provided by Prusa Support. Doesn't matter if printed in ASA or PLA but more pronounced with ASA.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 11

### Printer model

Prusa MK4 factory built.

u89djt, 2024-06-17:
Can you list the solves provided by Prusa Support? I'm just a fellow user puzzled by your experience. I've only seen that kind of gloopy extrusion on my MK3S when the filament is wet. Organic support works really well unless there's a wall count glitch, and I haven't found any in your slice yet. If it's not wet filament, I'm interested to learn what's causing it.

MartenKL, 2024-06-17:
Pretty sure it is some kind of overextrusion in the slicer for organic supports when placed on model, my filament is definitely very dry, always stored in vacuum and when printing always from a dryer. Also, filament type does not matter, changing support types resolves the issue.
/Mårten Kai-Larsen

MartenKL, 2024-06-17:
Decreasing temps, slowing down bridges and decreasing top gap to .1 mm on supports.
/Mårten Kai-Larsen

u89djt, 2024-06-17:
Let me possibly make a silly mistake, but would it not make more sense to increase the gap? In the extreme, something like 1mm would leave a physical gap and definitely not push upward. 
I'll slice it for my MK3S tomorrow and if the estimated mass of filament used is identical, I'll print it and weigh the object including the brim and supports. It'll definitely print OK for me, so there should be something to contrast. Maybe I'll just print half of one arm to save time and filament. 
If the estimated mass is different, but I can't see a difference in the preview, that would suggest that there's something odd in the MK4 profile? You could weigh yours if you haven't binned it.

u89djt, 2024-06-18:
I haven't been able to persuade the MK3S slice to come out with an identical support field so I think I can't try a meaningful experiment.

MartenKL, 2024-06-18:
I would like to see the print without the mid bulging using organic support.

u89djt, 2024-06-18:
Then I will print it! Would one half of an arm be enough? I'm absolutely happy to print as much as necessary.

MartenKL, 2024-06-18:
My guess is yes but time per layer and bed temp will be different
MvH
Mårten Kai-Larsen

u89djt, 2024-06-18:
Ah, you're doing this via the emails - I edited my response and added a 3mf file for the MK3S for you to add things to. You can drop other stuff in so you see what the supports (should?) look like in general and see for yourself how the settings do differ in places.

u89djt, 2024-06-18:
I'll run this after lunch (it's 1015 here) unless you post a specific 3mf file: 
[image]

u89djt, 2024-06-18:
Depending on how that goes, I can run it on the XL when it's free. I should be able to match speeds with the MK4 - I assume it's slower.
Obviously if anyone else is reading this and has a MK4 and some time, speak up :)

MartenKL, 2024-06-18:
I don’t use input shaping ever, prints don’t get good enough
MvH
Mårten Kai-Larsen

u89djt, 2024-06-18:
That sounds like we have another different experience - I'm enjoying the quality of input shaper prints on the XL. We could keep digging around for clues here. If you can post photos of the quality difference you see between an input shaper run and a non-input shaper run, someone might be inspired to suggest something new. 
Oh hey, what's your filament path to the printer? If it's struggling to pull filament through a twisted PTFE tube from your dryer, that could cause hiccups in the extrusion. It would also make the extrusion behaviour different to what the input shaper logic presumes.

MartenKL, 2024-06-18:
No snags on path, smooth as butter and too much other things to worry about that I don’t want to add IS to the mix. This is a slicer issue and has nothing to do with printer or material, I am certain.
MvH
Mårten Kai-Larsen

u89djt, 2024-06-18:
Aha, then we're both certain of different things - a healthy basis for exploration :) I'm going to build, post, print and photograph a test project that will show a range of organic support behaviours. I'm bound to learn things from the exercise, so it'll be worth it whatever the outcome. I'll post 3mf files for the MK3S and MK4 for anyone inclined to investigate.

MartenKL, 2024-06-18:
Well I’ve already printed the same model over ten times, removing organic solves the issue, nothing else.
MvH
Mårten Kai-Larsen

u89djt, 2024-06-18:
I'll go ahead and develop those tests and only come back when it's printed so you're not flooded with emails.
If there's a dev reading this and hanging back to spare my blushes, just go ahead and tell me I'm wrong if you know what's going on :)

u89djt, 2024-06-19:
TL;DR these organic supports are unlikely to fall over, so I want to recommend you try printing it. If the supports get knocked over, cancel the print, and we'll try something else.
[organic condition test.zip](https://github.com/user-attachments/files/15905297/organic.condition.test.zip)
[image]

In your print, most of the organic supports on the inside of the body have fallen over during the print. I've brightened the bits we can see in that photo.
[image]
I've attempted to simulate that situation by printing the right arm of your object and knocking over the internal supports during the print. I picked the layer where the top of the most obvious one on the left seems to have broken.
Here's the paused print at that layer:
[image]
And here it is after knocking down most of them, and plucking one from a screw hole:
[image]
The print proceeds from there with filament draping into the place where the supports should have been
[image]
Then we see some of the mid-air extrusions beginning to succeed somewhat - putting up a bit of a fight - because of the birds nest beneath them:
[image]
And amazingly, eventually, things become completely coherent before it gets to the top:
[image]
And the final object is apparently oblivious to the sabotage.
[image]
This print hasn't got the lifting of the edge or the rippling in the upper surface visible in yours, so it seems like the toppling of the internal supports doesn't explain those effects, at least not by themselves.

My guess is that there is some drooping in the middle, which pulls material away from the edge. More drooping due to less successful recovery of the support would exaggerate that. This is pure speculation - we'd need to look more closely at your print.

We've established that your object can print successfully with organic supports, even if it's sabotaged. That tells me there aren't any affecting slicing errors.

Organic supports get knocked over when there are excess or misplaced blobs of filament, or they have voids in fine structure. That happens if the filament is over or under diameter at some points, or if it's wet, causing it to spit, disturbing flow. Both of those features would also cause random change in pressure buildup and response, hence distorting the effects of input shaping.

Snug supports are a thoroughly self-reinforcing structure, and so survived this in your successful prints. They have full-height neighbouring support to lean against. I would expect to hear loud ticking when blobs are hit by the nozzle.

The rippling in the surface and the lifting of the edge of your print tells us that the filament hasn't deposited and stayed where it's supposed to be.

Unless you have a problem with your extrusion axis, I think (fellow user, not a developer or in any way affiliated with Prusa) filament quality is sufficient to explain what we see in your photo. Either the filament varies too much in diameter, or your local conditions mean that you need to put more time into drying the filament.
It took me a while to appreciate how much effort and care it takes to get the filament dry enough to be fast and neat. Vacuum bags don't actually exclude moisture entirely, they just limit the amount of air the filament sits in and slow down moisture transfer. Mylar-lined bags are impermeable to moisture. You need a bag of indicator desiccant in there, too.

After the sabotage, my print clicked and crunched as the nozzle hit stray plastic building up in drifts until it was kept in place by the accumulation of better and better approximations to the right shapes.
Your print will certainly have made those noises, and added clicks as the supports were knocked down.

FDM 3D printing generally whines and chunks mechanically as the servos drive and stop the axes very quickly, but good filament - a constant correct diameter and no excess moisture - doesn't make significant noise during a print.

I also printed this with some old wet filament (not bothering to knock down supports), but it didn't fail. It illustrates the sort of things you look for with wet filament: stringing and blobs,
[image]
and the tell-tale drips every time the nozzle touches the bed during levelling before the print
[image]
I can't see any of those in your photo.

Many filaments arrive disappointingly wet. Yes, they're in vacuum packaging with a desiccant bag, but they've generally been stored and shipped for a while. Unless the bag has a mylar layer, it lets moisture through slowly. And the desiccant bag has finite capacity, and you can't tell how wet it is if it's just one of those white printed ones.

OK, there's one email hopefully! It would be great if you could reply in the github interface so it isn't copied out in full, but see what you think. Cheers!

closed 2026-09-29 by prusaslicer-bot[bot]