prusa3d/PrusaSlicer #12840: 2.8.0-alpha5 - Improved G-code Viewer - Add data of the actual segment

Help3D-Padova, 2024-06-17:
In the lasta 2.8.0-alpha5 the new pop up panel is great

[image]

Is possible to add here 

[image]

the value of the actual parameter monitored, linke in Orca Slicer or Bambu Studio ? 

[image]

lukasmatena, 2024-06-21:
Speed and volumetric flow rate was added into the info dialog in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Does it solve the issue?

Help3D-Padova, 2024-06-24:
> Speed and volumetric flow rate was added into the info dialog in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Does it solve the issue?

Great ! Thank you !

Help3D-Padova, 2024-06-24:
Fixed

closed 2024-06-24 by Help3D-Padova