prusa3d/PrusaSlicer #12841: New project is greyed out

jleslie48, 2024-06-17:
Well maybe its a language issue, but when I first start the program and I want to start a new project, I find it very confusing to have "new project" greyed out.  The whole point of STARTING a new project is to, well,  start a new project.  There is no reason for it to be greyed out.   I want to import a 3d model, start working on it, and then save it as a project so I can work on it later.   What am I not understanding?     How do I start a new project where the first step is not to declare and name the new project?

P.S.  pet peeve:  if you grey out an option, why when I either click  the option or hover over it you not tell me the boolean logic reason its greyed out???   Don't feel bad though, 99.99% of all software is guilty of this same oversight.

jleslie48, 2024-06-17:
"When you open the software it automatically IS a new project."   That is non-standard and not intuitive.  Most software on starting is waiting for you to either start a new project or open an existing one.  You are also mistaken on your analogy with delete.  Delete is not waiting for you to add something to delete, delete is activated when you have something hi-lighted and is only greyed out when you don't have a part selected.

neophyl, 2024-06-17:
It may be non standard,  so ?  PS does lots of things in a non standard way.  I'm afraid that's something you will have to get used to.  Or not.  

I could also have chosen a better analogy but it was a quick reply while on a break at work,  the point still stands that on opening the software you are already in a new project state so there is no point having New Project as an option.

jleslie48, 2024-06-17:
I had a car once that moved the car horn from the middle of the steering wheel to the directional by pushing that in.   I couldn't get rid of that car fast enough.  Sometimes things become a standard for a reason and that's a good thing.   Bucking the norm for just for the sake of being different is not being original or an innovator just for being being different.

kubispe1, 2024-06-19:
Thank you both for your opinion.

closed 2026-09-29 by prusaslicer-bot[bot]