prusa3d/PrusaSlicer #12842: PVA/BVOH filaments incompatible with MK4 MMU3.

SveinHaDD, 2024-06-17:
### Description of the bug

Lots of filaments are available for MK4 but are incompatible with MK4 MMU3. The most important one is PVA/BVOH which is more or less unusable without MMU3 but when using MMU3 they are unavailable. I believe I can hack the filament settings myself but this should really be done the proper way by Prusa...
[image]
[image]

### Project file & How to reproduce

Easy to see difference by selecting "Filaments Profiles Selection"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.4+linux-x64-GTK3

### Operating system

Linux Mint 21.3 Cinnamon

### Printer model

MK4 MMU3

mw46d, 2024-07-02:
I ran into the same issue but I took the MK3-MMU files and created versions for the MK4-MMU wich kind of work for me. They are probably far from perfect, but maybe, they can serve as a starting point for others?! 
[image]
[image]
[PVA.zip](https://github.com/user-attachments/files/16073951/PVA.zip)

SveinHaDD, 2024-07-03:
Thanks but I tried to install these configurations and they messed up the slicer menus and still don't allow for access to PVA/BVOH when using MMU3...

mw46d, 2024-07-03:
:-( I'm using 2.8.0. on Linux and things seem to work just fine:-(  You added them as an user-preset Filament and two 'Print Settings'?  
[image]

unclej84, 2024-10-04:
I created an issue for this [here](https://github.com/prusa3d/PrusaSlicer-settings-prusa-fff/issues/1)

mjablon1, 2025-01-15:
Hi Prusa, can you make this available or at least comment why PVA/BVOH is not compatible with MK4?

closed 2025-02-12 by rtyr