prusa3d/PrusaSlicer #12846: More detailed extruder clearance configuration

logiclrd, 2024-06-17:
**Is your feature request related to a problem? Please describe.**
I have a Cocoa Press chocolate printer. Its print head is unusually large and deeper than it is wide. I would like the ability to better capture this shape in the configuration of extruder clearance for sequential part printing. The current configuration system only allows me to specify a radius, and because of the shape of the print head, I must specify a very large clearance value to account for just one direction. A lot of area is excluded that doesn't need to be.

**Describe the solution you'd like**
I'd like the ability to configure extruder clearance separately for left, right, forward and backward from the nozzle.

**Describe how it would work**
In the configuration screen, there would be four input boxes next to "Extruder clearance - radius" and checkbox marked something like "Same". If the checkbox is checked off, then only the first input box is enabled, and entering a value into it would immediately copy this value into the other four boxes. When unchecked, the four input boxes can be given values independently.

Then, when computing the space in which collisions must be avoided, a shape is formed by extending a rectangle out in each of the four directions specified by these separate input boxes and then rounding each corner by the minimum of the two radii in that corner.

When calculating clearance with parts, the order of printing is taken into account, and the radius must only be kept clear of previously-printed parts. In my case, where there is considerably more print head further back, printing front parts first would minimize the needed clearance.

**Describe alternatives you've considered**
There is no alternative. The only work-around I've found is to specify a radius that encompasses everything that makes up the print head, which means that a large amount of empty space is also excluded.

**Additional context**
This shows the current options for the exclusion zone (top down, with the cross indicating where the nozzle itself is located):

[image]

The small circle doesn't work because the part of the head that is further back can collide with previously-printed parts. But the large circle excludes far more area than it actually needs to.

This shows a more ideal version of the exclusion zone:

[image]

To configure this, it would need to be possible to specify the amount of exclusion needed in each direction separately.

neophyl, 2024-06-17:
This looks a lot like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7787

logiclrd, 2024-06-17:
@neophyl I'd say you're right, though this issue has a lot more detail in it. :-)

closed 2026-09-29 by prusaslicer-bot[bot]