prusa3d/PrusaSlicer #12847: Prusa Connect randomly does not display previews sent by Prusa Slicer

zbkingCZ, 2024-06-17:
### Description of the bug

Hello, 

Well, I don't know if it's a PrusaSlicer problem or Prusa Connect. After sending a file from PrusaSlicer to Prusa Connect, previews with printed objects are not displayed randomly. If the preview of the printed object is not displayed, the printing plate is also not displayed and therefore the "Cancel Object" function cannot be used. The "Cancel Object" function can only be used manually on the printer.
[image]

Best regards

Zbyněk Král

### Project file & How to reproduce

[Tesla Label V01_0.4n_0.2mm_PLA_MK4IS_1h56m.zip](https://github.com/user-attachments/files/15874408/Tesla.Label.V01_0.4n_0.2mm_PLA_MK4IS_1h56m.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64, PrusaSlicer-2.7.4+win64-202404050928

### Operating system

Windows 10 (sestavení 19045), 64bitová edice, Total RAM size [MB]: 8,510MB

### Printer model

Original Prusa MK4, Firmware 6.0.2+14873

zbkingCZ, 2024-06-17:
Um, just a little note. If I send a print file from Prusa Slicer to Prusa Connect in .bgcode format, everything seems to be fine.

[image]

kubispe1, 2024-06-19:
Hi,
Thanks for reporting this issue. It's most likely a problem on Connect's side. I'm forwarding it to the team for investigation.
- In the meantime, I can confirm that I've observed this issue as well. Several gcode files aren't displaying thumbnails correctly. However, it's good to know that bgcode files are working as expected.

dunkmann00, 2024-06-19:
I've not used PrusaConnect, but have experienced something that might be similar to what you're describing with PrusaLink. In particular, the fact that it only happens with text gcode and not binary gcode. 

If this is the same thing these might be related:
- [Thumbnail generation not parsed by PrusaLink](https://github.com/prusa3d/Prusa-Link-Web/issues/475)
- [Increase the lines searched for PNG image](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3954)

zbkingCZ, 2024-06-19:
Thank you for answer. I switched the slicer to .bgcode format because the "Cancel Object" feature in Connect is more important and by the way it's just brilliant 👍.
However, on the other hand, I really miss the direct editing of this format in text mode. Sometimes it is necessary for me to directly modify the source code for the printer.

Printer606, 2024-06-23:
I am also experiencing these issues, except I can't get them to show up at all.

kukulin, 2024-06-26:
I have the same issue in in version 2.8.0-RC1 win64

kubispe1, 2024-07-11:
If I remember correctly, there was a tiny issue with cookies on our side, which we fixed in one of the recent releases. It should be much better now, but I noticed some missing thumbnails. I assume it is an issue with the communication between Connect and Fw, as I see the same problem in the slicer webview and in common browsers (previously, this only occurred in Slicer).

I am closing the issue as resolved on our side. I have also forwarded the remaining issue to the Connect team.

closed 2024-07-11 by kubispe1