prusa3d/PrusaSlicer #12849: 2.8.0-alpha5 Time Remaining til Color Change No Longer Showing on Printer Screen

nathan118118, 2024-06-17:
### Description of the bug

the MK3S+ screen used to display total remaining print time and time remaining until next color change, and it would change about every 5 seconds. Now it seems to only showing total print time remaining. I had a color change scheduled to occur, but since it never showed time remaining until color change, I assumed I accidentally forgot to put in the color change. But nope, it ran the correct color change procedure.

### Project file & How to reproduce

Add a color change to a project and view MK3S+ screen. Never shows time remaining until color change, only total time remaining.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Windows 11

### Printer model

Prusa MK3S+

nathan118118, 2024-06-18:
It looks like adding a pause still triggers the time remaining on the screen,  but not clicking the plus to add a color change.

michael-gebis, 2024-06-29:
I'm still seeing the same thing in 2.8.0 (the actual release)

Looking at the generated .gcode, I see M73 Dxx codes, but not M73 Cxx codes. So it looks like the "time remaining to next stop " codes are being computed and inserted for stealth mode, but not for normal mode.

grimbeaver, 2024-07-15:
This is an issue in the released 2.8.0 when slicing for the MK3.5 as well.

grimbeaver, 2024-09-19:
Have not tried this yet in v2.8.1 but since this bug has not been resolved yet guessing it did not get fixed.

nathan118118, 2024-09-19:
> Have not tried this yet in v2.8.1 but since this bug has not been resolved yet guessing it did not get fixed.

I haven't tried rc1, but it looks like a related bug was fixed....so maybe this one just didn't get closed. Crossing my fingers! 😁

nathan118118, 2024-09-20:
> Have not tried this yet in v2.8.1 but since this bug has not been resolved yet guessing it did not get fixed.

Nope, still doesn't work. Now instead of showing incorrect color change times, it simply shows time remaining, and never toggles back and forth to the color change time remaining.

wkautter, 2024-09-26:
Loaded v2.8.1 for MAC a few days ago and used for the first time today. Having the same problem of no "time to color change" on display of MK3.

CodeKlopfer, 2024-11-07:
I can confirm no time till color change is displayed on MK3S+ with gcode produced with "Version: 2.8.1+linux-x64-GTK3
Build: PrusaSlicer-2.8.1+linux-x64-GTK3-202409181354"

puterboy, 2024-12-03:
This seems like a simple bug... why hasn't it been fixed yet?
And if it is a planned regression, shouldn't that be in the release notes????

nathan118118, 2024-12-03:
> This seems like a simple bug... why hasn't it been fixed yet? And if it is a planned regression, shouldn't that be in the release notes????

Yah, not sure. I looked through the 2.9 alpha notes and didn't see this mentioned, but I haven't tried it yet, so not sure if it has been fixed.

puterboy, 2024-12-05:
Given that this is a regression that breaks a very important and useful feature, is there any way of getting Prusa devs to pay attention to this?
It drove me crazy yesterday as I had a multi-color print that required filament changes every few hours and it would have been *very* helpful to be alerted when to come back to my printer...

puterboy, 2024-12-05:
That's great to here! Thanks!

lukasmatena, 2024-12-19:
Correct, this is already fixed. Closing.

closed 2024-12-19 by lukasmatena

hraftery, 2024-12-19:
For posterity note that it seems the Prusa-Firmware bug mentioned above was **not** the fix. It was determined to be a PrusaSlicer bug instead, fixed in https://github.com/prusa3d/PrusaSlicer/commit/a62f846e0bd18f96947528f190f2925cff984f60 and https://github.com/prusa3d/PrusaSlicer/commit/a62f846e0bd18f96947528f190f2925cff984f60 . These changes are included in [2.9.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-beta1) and mentioned in the release notes:

> Fixed export of M73 C lines to G-code to fix displaying remaining time to color change. This was partially fixed in 2.9.0-alpha1 already, but not in all cases. https://github.com/prusa3d/PrusaSlicer/pull/4734, https://github.com/prusa3d/PrusaSlicer/issues/12849

The link for #4734 should probably be https://github.com/prusa3d/Prusa-Firmware/issues/4734 instead.