prusa3d/PrusaSlicer #12850: Sinusoidal truss infill

laukejas, 2024-06-18:
**Description**
Several closed-source, proprietary slicers have the option to generate a sinusoidal truss infill, typically used for parts printed vertically that have more or less consistent thickness. This infill is geometry-aware, meaning that it always tries to keep the same angle to the perimeter walls in the X/Y direction. In the Z direction, this infill shifts in a sinusoidal wave. Here are a few examples for reference:

[image]

[image]

[image]

**Benefits**

This infill is typically used for printing large and lightweight structures very quickly. When printed with a single perimeter, it allows printing with a constant toolhead and extruder speed and zero retractions, similar to Spiral Vase mode, which can be highly beneficial for lots of filament types, reduces the risk of clogs, failures or print defects, and results in a very weight-efficient structure.

**Slicer integration**

This infill could be included with the rest of the infill types, with some setting specific to it:

1. Truss angle - the target angle between the neighboring truss segment, default at 90°. Alternatively, this could be indirectly controlled with fill_density, as higher density would result in higher angles.
2. Sine wave amplitude and frequency - determines the shape of the truss shift in the Z direction.
3. Truss and sine wave offset - to allow the user to specify where the truss starts at the first layer, to allow lining up truss infill for parts that will be glued together.

Alternatively, if the requirements of this infill type are too inconsistent with other infill types, it could be treated as an Experimental Mode, or something similar to Spiral Vase (setting that adjusts multiple other settings to make it work).

**Software implementation**

The following is just a suggestion of how I'd do it. There might be a better way.

1. For each layer, slicer should run a thickness analysis in XY direction, and determine two opposing perimeters that are closer than the other two. In the following picture - the blue lines:

[image]

2. Pick an arbitrary point on one of the perimeters near the end, and determine a normal direction from that point (yellow):

[image]

3. Extend a ray from that point at an angle that is half the user-specified truss angle, until it hits the opposing perimeter (green):

[image]

4. Repeat steps 2 and 3 to bounce the rays back and forth between the perimeters, until the last ray no longer hits the lines of the main perimeter pair. At each reflection point, an average between the user-specified truss angle and the perimeter normal direction will have to be calculated, to avoid the truss from developing a sawtooth pattern.

[image]

5. Generate infill lines on the rays.

6. Move to the next layer, offsetting the ray deflection points in the X/Y direction to account for the change in part shape and the sine wave shape.

**Final remarks**

It is likely that this infill won't work for all parts universally, for example - parts that don't have obvious "thickness" to them, or it is extremely irregular. However, it should still work reasonably well for parts that have a reasonably consistent thickness, even if the opposing perimeters are not strictly offset from one another.

murk-sy, 2024-06-18:
I think there are a lot of cases where this sort of infill would be preferred even in normal prints, sort of like a long-thin-section version of "small perimeter" option. Having an option to set it as percentage of nozzle width would be nice for normal prints (i.e. maximum width 300-600% of nozzle width).

This is an extreme example (gyroid, with 0.6 nozzle), but the infill is completely missing for over 1mm of height here

[image]

And the parts where the infill is present are likely unnecessarily slowing down the print

[image]

murk-sy, 2026-09-24:
Bump, still relevant as of 3.0.0-alpha12

[image]