prusa3d/PrusaSlicer #12853: Sign in to Prusa Connect in xfce does not work

ufgum, 2024-06-18:
### Description of the bug

xdg-open does open the prusa slicer, but the login credential seems not to be passed.

[image]

### Project file & How to reproduce

xdg-open does open the prusa slicer, but the login credential seems not to be passed. The Log in stays at it was before.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0-alpha5+linux-x64-GTK3

### Operating system

arch linux - using appimage

### Printer model

XL IS 5

kubispe1, 2024-06-19:
Are you also encountering issues when transferring models from Printables to PrusaSlicer? 
[image]

lukasmatena, 2024-06-21:
Can you please retest in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1) and let us know? It embeds the login dialog into PrusaSlicer's own window, it should bypass whatever the problem you are having.

ufgum, 2024-06-21:
> Are you also encountering issues when transferring models from Printables to PrusaSlicer? [...]

Sorry, I did reply by mail.
No, this is working.

And after pressing the login in the browser, the prusaslicer opens, but no credentials are passed or received.

ufgum, 2024-06-21:
> Can you please retest in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1) and let us know? It embeds the login dialog into PrusaSlicer's own window, it should bypass whatever the problem you are having.

Hi,
Sign in to Prusa Connect works fine now.

data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAYMAAABBCAIAAAC8UV9OAAAACXBIWXMAAA7EAAAOxAGVKw4bAAAS0UlEQVR4nO3df1RT5R8H8M+2u41tbJPxG8EUELC0BBE4mn2t/BVhilgaikfSNPUYaYBo/igrEBURRUNTUUktPRip53RCQkVUFBFEMdAJqEMQlDEY22Dbvd8/Lq01BiGg9OPz+mt7uPe5D/du7z3Pc+92GV999RUghFCfYvZ1AxBCCJMIIfQ3gEmEEOp7mEQIob6HSYQQ6nuYRAihvodJhBDqe5hECKG+h0mEEOp7mEQIob5HmC3VarVS6T2lUsVisZ9zgxBC/0pabYtIZOnu/gKbbSZVzCSRVqu9evW6TqdXqZpaWlQURT37RiKE/uUsLAQtLZqGhgZf31fah5GZ0ZlUek+r1dXX12g0zRhDCKFeodE0y+WPdDq9VHqv/V/N9IkaG5VqtfLZNwwh9J+jUjUplbz25Wb6RGw2V6NpfvZNQgj957S0mJ99xnNnCKHnp6MJHwJnghBCz1n72ME+EUKo75m/nggh9J/lzte9atXqJdAKie4MmFR6xi0l+5ycU656injBJEII/SHIVvPxCz09df6KUPu+o2r7fcuTtRZdXAWTCCHUxpWn/fgFZbOekfaQX6kmmnRP1ydiMNoe+Ag1Hzi3Lh2gvNbIrtKwurIuJhFCqM1oKzUApD3kH39k5pKfrrvdzNGQ6sUDmqfYaXbeF3RlFZyxRgi18RC0AkCOnNvzqvIbOQDQn6vv4vKYRAihzthyupQmpos95WQ3js4QQgYcAAVpdLFPnFejX7/WKg1rZamwSmM+Lpwt9JtfVNhyyLN13C/vCulCNckAALLLG8Y+EULIvP4Wer9+rfSDzZ61/S107ZcxxBAAjLFqseF0PXz+BJMIIWRelYZlOPNlzTYTRv25OkMMAUC5injc2s1IwSRCCHVoZanovtJ8GPXn6ja71xpiqEbFXC0VdXtD/6QkYrPZ//vf/wgC57aeNw6H09dNQH2jSsNaK7WsU7cC8AB41mzOFs/G/lx9f65+x9AmB0sLvQ70OqhTM2Nui7vdIYJeSaIDBw5YWHT1Ssqe4HA4S5YsMfm1N2dn566su2DBguPHjx8+fDg6OjogIODZNBBCQkJ27dr1jCo3wefznZycelIDj8dbuHBhUlLSqVOnjh8/HhwcbHYxe3v7b7755mkrHzhwYHp6+vHjxxMTE2fMmMFk/pM+85CxBxp2VJmtXNt2BK3Y5I6hjTuGNop//y5IvZYZVSZ+0LUrGDvSC68PFxeXnlfSFRRFkeSf5sNEIlFkZGRX1nVxcUlNTQ0NDc3MzExISLCxsel5exiGS0p/N2PGDAAYOHBgzyv/S0OGDAkJCelJDV9//bWNjc2+ffuCgoIWL16cn59vdrFHjx4tWrSoG/XLZLKQkJD4+HhfX9+OYg79IzzQsKPLRIYwEhNU78YQ9MpZfIIgRCKRRqMxlLz11lshISGWlpbFxcXx8fF6vb5fv37h4eGjR4+WyWQNDQ2HDh26c+eOm5tbeHj44MGDCwoKHBwcVq9erVKpJkyYEB4eDgB5eXlJSUkA4OXlNXv27MGDB9+6dctkuytXrvTz80tMTMzJyTlx4sQbb7wxffp0sVhcUlISFxen05mZ6i8qKrp8+fKoUaNOnz4dGxtbWFg4derUTZs2eXt7V1RU/PzzzwCwePHivLy8a9euvfvuu+PGjePz+Q8fPoyJieHxePHx8SKRSK1W79+//8qVK8Y/buDp6fngwYOzZ88GBwcnJibShWFhYdOmTWMwGGlpaenp6QEBAUuXLhUIBLdv346OjnZ2dg4JCfH39ydJcv369bdv33ZwcIiMjKTj1d7eftWqVREREf7+/uPHjxeLxc7OztXV1bGxsQwGY9WqVZaWlq6urikpKWVlZfTmhgwZsnjxYpFIVFtbu23btgcPHowcOXL+/PlcLpckyc8///z+/fv0kt7e3l5eXsHBwS0tLQBQU1NDl7NYrA0bNtjZ2Wm12mPHjv3yyy/29vaxsbHz5s0bN27c0KFDt27dCgBjx4719PTctWuXv7//Bx98wOVydTrdunXrqqqqTHZ4TU1Nbm7u2LFj09PTExIStm/fXllZCQBJSUkJCQlVVVVJSUklJSVjxoyRy+VpaWl5eXmOjo7Lly+PiooCAAcHh5iYmGXLlrm4uERHR/P5fPooFBUVdefFirrrnoaILhNt9Gy0Yv/RG1DoGFFlop7HEDyL64kGDhw4bdq0VatWNTU1ffbZZ++9996RI0dmz56t1Wpnzpzp6em5Y8eOH374AQAiIyOPHTu2du3awMDAiRMnAoCdnV1MTMz06dPlcvnmzZunTp2akZGxaNGin376afXq1ZMnT3711VcNG9LpdCkpKU5OTsuWLQOAQYMGzZw5c/Xq1XK5fMWKFbNmzTpw4IBhYblcTg/rfHx8fH19ExISAMDPz6+goGDatGkAMGbMGMMYUygUCgQCNps9Z86cKVOmkCTJ5XIpilKr1StWrFCpVC+++GJUVNSVK1eM//Fx48adOXMmJycnIiKCTqIpU6aMHTv2ww8/fPz4MUEQEonkyy+//PTTT4uLi+nZrlmzZt29ezc0NNTHxychIWHWrFlMJtPe3p6ukM/n0495PN6YMWOWLFkilUpjYmImT56cmpp65MgRR0fH7du3G7ehsrKSbmFoaGhgYODu3bvDw8M3bdoklUq5XC4dOjRfX9+LFy/SJba2thwOh8FgyGQyvV6/bt06lUrl7Oy8ZcuWzMxMBoPBYrEAwMLCQiBou3JfIBBYWVkxGIzw8PD4+Pjy8nKT+nk8nl6vpyjKwcEhKCgoLy8PAMRisaEGsVhM73AnJ6erV6+GhYW9/PLLy5cvz8/PZ7FYdnZ2hp1gZ2dHUVRoaGhGRkZWVhaHwzH7GYOetRaqbRig1beFkU7P0PbS75v1/ug9ICCgqKiotrZWrVZnZ2ePHj0aAPz9/U+dOqXT6UpKSh4+fAgAQqHQxsbm4sWLJEleunRJpVIBgIeHR01NTWho6JIlSwDA3d1dLBbb2dldvXoVAC5fvqxWq423xWAwuNy2K9OHDx9+8+bN6upqjUaTlZVFb9dYREREenp6eHj4ihUr6I9uiqK+++47ADA7i6HVavPz83fv3h0YGEi/xyiKcnNzCwoKev311x0dHU2WpwcgPj4+Mpls8uTJAODm5paVlfX48WMA0Ol0bm5ud+7cKS4upp9yOJxXXnmF7oXRe8zV1bWjvVpRUSGVSgHg7Nmzbm5uAMDhcNo3W61Wu7m5zZgxw8fHZ9CgQRRFZWZmxsfHh4aGmgxsWSyWTCajH8+fPz82Nnbjxo30C83FxWXSpEmBgYHW1tZ0BnWEoqjTp09v3Lhx5syZJvUDwIgRIw4fPrxmzZqcnJzU1NRO6snJydHr9YWFhQqFoqOJv19//XXZsmULFixgs9ntt4WeNQeuftsQRT/iz9MjBLXFU+HQ5a90dKL3k8ja2trwuKmpqampCQCMp5nlcjkAGL/E1Wo1nUQAQJJkbm5ubm7u3r179+zZw2KxtFotPfQjSdJ4DNjJdpVKpVJp+ssGSUlJISEhS5cuvXTpEl1iGEqYvQETAKxfv37nzp3jx49PSkpiMBihoaHz58+vr68/c+aMyZLBwcFVVVWDBg0aMWJEVVVVVybFmUymlZWVcUkn/92jR4/+skIA2LhxY1BQUFlZ2bVr1+iSjIyMqKgoW1vbgwcPGs/o3bt3z8PDg34cFxcXFRVFZ/rEiROjoqJUKtXp06e70vVIT09fsWKFk5PT/v37+/fvb/yngoIC+kMlLS2NJMn202pPJT8/f+7cuSqV6tChQ8OHD+9JVehpOXB0xjHUqGM06tqOpg2H7JUwYlI9BgANDQ2Gp7du3fLz86OHM2+//XZ2djZFUbm5ub6+vhRFSSQSDw8PiqLkcnldXd3gwYMpinrppZdsbGwoiiotLbW3t6+pqSksLCwpKZHL5fX19RqNxsvLi6IoV1dXiUQCACZb5/F49HZHjhzJ5/MpigoMDMzKyjJZzGRFkiTphWmVlZWurq4URQkEgtGjRxvKCwoKPvnkkwEDBnC53FGjRu3cufPChQvDhg0z2QlTp05NTk7etm3b5s2bV61a5e/vb2dnJ5VK33zzTTabTTdAKpW6u7sLBAL6aUtLS1VVFb0tV1dXJpP522+/1dfXczgcOqYnTZpkaDOT2XakCIKgH6hUKroqAy6X6+npuWXLlsLCwmHDhtHrEgRx9+7drVu33r5928nJybBwYWGhh4fHtGnTjHcRRVGvvfbanj17zp075+rqSh9Ew59qamros3UEQRjaRhCEVCpNSEiorKx0cHAwqc3EzZs36Z0skUg8PT0NS/r7+1MUZW1tLRKJZDLZkydPuFyuyU7gcDi1tbVpaWkZGRlDhgyhKMrX19fsVlC30V8WMym0Z2sT3GsNMdSgZUaWiiJLRcZhlOCpsOfoTWsDAKqrB6h35olOnjxJj18OHTr0/fffe3h4HDx4UKlUFhYWnj9/HgCOHj0aERExduzY+/fvP378mB5W7N2796OPPmpubi4vL6e7MHV1dWvXrv32228fPnxoYWGRlJRUUFCwc+fOhQsXSiSS8vJyemRnUFVVVV1dvW/fvmPHjqWnp3t5eaWmpiqVyuLi4nPnzhkvyeVyGxsbO2o/k8nMzs6Oi4s7duxYa2trUVERg8EQCoUpKSkymUwoFJ48eVKtVp86deqLL75oamoqLS2tra01rO7k5MTn842njX788cfp06fv2rXLz88vIyOjoqLiwoULhw8fTk5OPnr0aEVFhUqlioqKWrt2bWJi4ty5cxUKRXJyMgBoNJpz584dOXIEAK5cudJJLyk/P3/OnDkpKSk7duy4ceMGvW52dvaBAweampoKCwudnJxYLFZcXJyVlVVLS0t1dTU9WUOrrq5etGjR559//v7776tUKpFIVFdXBwAnTpxYtmzZvHnzKisr6allNptNX09UVFSkUCiOHj1Kt41OzA0bNojF4tbWVplMRg+iaSRJUhTFZDKNR1KZmZnr1q0LCwtTKBTXr1+n+1xKpdLb23vq1KkNDQ2pqak6nU6v158/f954JzCZzKVLl/r4+DQ0NDQ3N69Zs8bKyio5OXnChAkKhaKjXYR6rh+b [... cut ...] BkIIdabzGHraxTqCSYQQalPWzAGAMVZ/cfKhK/zEWgDo4o09AH89FiFkcKGBN6e/do6TCgDuq1mN+qe8y9DvD3xELfOcWwHgJ7zfGULoaZWr2En3BBEvNC8e0Nzz2rbfE2CfCCHUHadqLUqVxBirVi9LnZDVna+3NusZvzUTZ+vxbtQIoR6Qqgjp04RIr8AZa4RQ38MkQgj1PUwihFDfM59EPfwpGYQQeipmkogkdRYW/OffFITQvx6PJ9DpzFw5aSaJBAI+n9/9O6ghhFBHeDxLodCyfbmZJHJzcyEIlkRiz+MJnn3DEEL/fgwGg8cTWFs7stlsNzcz9yUjTH7LBgAIgvD2Hlpe/qCpycLGpn/7dRBC6GmRpFYg4Lu6utA/Z2jyV/PXL7HZbE/PDu8zgRBCvQvP4iOE+h4mEUKo72ESIYT6HiYRQqjvYRIhhPoeJhFCqO9hEiGE+h4mEUKo7/0fz2XIOSHabOIAAAAASUVORK5CYII=

ufgum, 2024-06-21:
But on closing and opening prusaslicer, the log in has to be done again. The credentials got not saved in prusaslicer.

Even is an instance running and a second one opened. The second one is not logged in.

kubispe1, 2024-07-11:
The credentials should be keep in PrusaSlicer in Final 2.8.0 even on Linux.
Can we close this issue ?

closed 2024-07-24 by ufgum