prusa3d/PrusaSlicer #12857: dynamic speed for overhangs causes slow layer when there are no overhangs

Racemade, 2024-06-18:
### Description of the bug

If I have activated the dynamic speed controlled in the slicer, these lines appear on the 3D model, which show me that the speed is being throttled.
However, there are absolutely no overhangs.
As soon as I disable the dynamic speed, the lines are no longer there.
As soon as I move the 3D component by 90 degrees, they are also no longer there.

With activated dynamic speed
[image]

With deactivated dynamic speed

[image]
[image]

Expectation:
Dynamic speed remains disabled because there is no ov
[image]
erhang.

### Project file & How to reproduce

[Testdata.zip](https://github.com/user-attachments/files/15891557/Testdata.zip)

Go to Printer Settings -> Speed -> Dynamic speed for overhangs -> active dynamic speed

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows

### Printer model

Prusa XL

SachCZ, 2024-06-18:
Hello @Racemade, this issue is already resolved in the latest alpha, please try it using PrusaSlicer-alpha5. Thank you for your report.

closed 2024-06-18 by SachCZ

Racemade, 2024-06-18:
Hey @SachCZ ....Sorry! My bad. Haven´t seen that there is an alpha. Was just reading the description and hell yeah!
--> Slowdown on overhangs was incorrectly applied to non-overhanging parts of the extrusion.

Merci!

SachCZ, 2024-06-18:
You are welcome! :rocket: