prusa3d/PrusaSlicer #12859: Change Extruder Order on PrusaXL

TechnikTim, 2024-06-18:
See also:
https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-hardware-firmware-and-software-help/extruder-print-order-multi-tool/#post-683733

If you want to print with multiple colors or materials, you currently have to sort the colors or materials on the printer so that they are printed in the correct order. This is very annoying in the long run and leads to errors.
It would be easier if you could tell the slicer in which order the extruders should be used.
So instead of always 1, 2, 3, 4, 5, you set the order in the slicer to 2, 4, 5, 1, 3.
This is particularly important if you want to print text or small details on a single-colored background. It then makes more sense to print the background first. To do this, however, the color of the background must be on an extruder whose number is smaller than the extruder with the color for the text or small details.

The whole thing gets even worse if you want to print with different nozzle diameters in the future. Then you can no longer simply swap colors, but have to change the nozzle and recalibrate the offset. This means that a 20-minute print quickly becomes a 2-hour project.

I think and hope this would be easy to add.

wafflecart, 2024-06-19:
I think you might be approaching it wrong, you CAN print any area with ANY EXTRUDER but you have to do one of these:

- If its an stl and a single part/body and you cannot change this (i.e. you didn't design the model) you can use the multi material paint tool in prusaslicer, paint any area with any extruder.
[image]

- If it's your own model, design your model with multiple bodies/parts and with 3mf format, e.g. in fusion 360 make sure to do each bit in separate bodies, but instead go to the main component and export as 3mf from there. When imported into prusaslicer it will be together but listed as multiple parts, you can select which part and which extruder to use.
[image]

u89djt, 2024-06-19:
Have another look through the articles about the menus on the XL: https://help.prusa3d.com/article/tools-mapping-xl_668277

TechnikTim, 2024-06-19:
Thank you for the answers.
I think you're right.
I always assigned the colors in the slicer the same way they are loaded into the printer.

If I assign the colors in the slicer according to the print order and then swap the extruders on the printer, it should work.
So if I want to print with black and then white, for example, I set black to extruder 1 and white to 2. Before printing, I then say that black should be printed with extruder 3 and white with extruder 5.
Then the order and the extruders are correct without me having to reload filament.

Nevertheless, this is more of a workaround than a solution. I have to reassign the extruders on the display every time I start a print. Better then reloading all Toolheads but still not great.

A simple option in the Slicer where you can set the order would be easier.

u89djt, 2024-06-19:
One more thing: you can drag objects to new positions in the list. Might that be the missing ingredient? With that, and the fact you can click on the right of each object and part's tool indicator to select from a drop down list, I think you have everything you need, but I may not be understanding what you mean.

u89djt, 2024-07-31:
I enjoy trying to dig out workarounds - can you post an example model that fits your problem category?

u89djt, 2024-07-31:
That one works for me: 
[image]
[XL5T 1+Lizard+4+Color.zip](https://github.com/user-attachments/files/16442136/XL5T.1%2BLizard%2B4%2BColor.zip)

petternilsen, 2024-07-31:
Well, yes, but this is such a needed addition.  We have to just admit that the bambu 3mf format is the most common, and prusa will need to adapt to it at the best possible way, meaning a way to remap the colors. The remapping is already available when loading a multiple STLs as a multi color model, so the code is there, it just needs to be applied for this particular use-case, a very common use-case.  Interestingly enough, there are similar issues filed in the bambu lab github, so it's a fairly common request.

petternilsen, 2024-08-01:
> Well, the code is there to attach a label to a complete object. You're asking for association of an attribute with some triangles on a mesh. There are myriad ways that might be implemented. That there are open issues on both sides of the fence tells us that there's either a technical barrier right now, and/or that it has been recognized that the current scheme will be limiting with respect to future features that mean nailing parts of it in place with further integration is asking for trouble. I imagine there's a lot of work and conversation going on behind the scenes. I would speculate that unloading the wrong filament and loading the right one is regarded as a lightweight task fundamental to printing different things. If it's OK for me to ask: what dissuades you from changing the filament loadout? You presumably want a developer to work on this, so they'll need motivation if they're not already doing it. I'm ignorant of the problem, as each of my prints involves a different combination of filaments.

My main motivation would be that I do have many different color models, and re-loading a new set or changing the filaments around to fit the model, just gets old really fast.   On the other hand, I'm motivated and experienced enough so I've grabbed the code and will have look and see how hard it would be to implement.  Since you mention "triangles on a mesh", they already must have color information and color filament offset stored, so it doesn't seem like a hard task to just map those to a different value.

u89djt, 2024-08-01:
I wish you a following wind.

foorschtbar, 2024-09-06:
Back to the original Request: I also had the problem, that the printing color order comes from the order of the heads used. I will now change it in my printer from lighter to darker colors, but a option to select the order per layer would be great.

closed 2026-09-29 by prusaslicer-bot[bot]