prusa3d/PrusaSlicer #12860: Failed to Load image (from temp drive)

FoldZero, 2024-06-18:
### Description of the bug

[image]

Happened straight away on first launch adn subsequant launches..
Installed PrusaSlicer-2.8.0-alpha5+win64-202406142204

### Project file & How to reproduce

no project related
Failed to load image from file "C:\Users\X\AppData\Local\Temp\prusaslicer-avatar-10681608104833421399.png".

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0-alpha5+win64-202406142204

### Operating system

Win 10 latest

### Printer model

cr6

Jan-Soustruznik, 2024-06-25:
Hello, @FoldZero  Could you share your avatar, please? We need it for investigate.

FallenGameR, 2024-06-27:
I have the same bug in 2.8 release. Tried to resave the avatar file with different programs, the error stays the same - it can't read the png file. Attaching the avatar file.
[image]

lukasmatena, 2024-06-28:
Thanks. We managed to find and fix the problem. It was not in PrusaSlicer, but on the server side. The fix was already deployed, so this issue should no longer show in any PrusaSlicer version.

Avatars from Prusa Account showed correctly, the problem appeared when avatar from Google account was used.

Can you guys confirm that the issue really disappeared?

lukasmatena, 2024-06-29:
Closing. The fix was confirmed in many of the duplicates.

closed 2024-06-29 by lukasmatena