prusa3d/PrusaSlicer #12861: Cannot build 2.8.0-alpha5 on Linux

jernejp21, 2024-06-19:
### Description of the bug

When trying to build PrusaSlicer 2.8.0-alpha on Ubuntu, I get errors about wxTimerEvent has not been declared in PrusaSlicer/src/slic3r/GUI/UserAccountCommunication.hpp. Build process works fine on 2.7.5-rc1 version.

### Project file & How to reproduce

I tried with clean git clone. Followed procedure for building dependencies and PrusaSlicer, as described in doc/How to build Linux et al.md. I used the same commands as in the document.

Screenshot of errors:
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Ubuntu 22.04

### Printer model

Not relavent

SachCZ, 2024-06-19:
There are missing includes.
Add:
`#include <wx/timer.h>` in UserAccountCommunication.hpp and `#include <boost/algorithm/string/split.hpp>` in UserAccountCommunication.cpp.

C++ includes are hard. We compile the software on 3 different platform with different compilers and see no issues, but still this happens quite often for other compilers/compiler versions.

closed 2024-06-19 by SachCZ

jernejp21, 2024-06-19:
@SachCZ Thank you for fast reply. I know how is, when it's working on your local machine and you forgot to include something in the commit.

jernejp21, 2024-06-19:
@SachCZ I have tried adding includes, but now other things are missing.

[image]

SachCZ, 2024-06-19:
Hi, this is another missing include. Something like `#include <wx/webview.h>` in WevView.cpp.

staal54a, 2024-06-23:
I just tried building 2.8.0-beta1 on Ubuntu 22.04 and I got the same wxWebView errors as above (the header files called out above were already included). I was able to clean, checkout 2.7.5-rc1 and it built successfully with no changes.

jernejp21, 2024-06-23:
@staal54a
Please see following PRs:
- ~~#12855~~ (already in master branch) 
- #12868 
- #12869 
- #12886 

You can do `wget --output-document=patch1.patch https://github.com/prusa3d/PrusaSlicer/pull/12855.patch` and then `git apply pathc1.patch` to apply changes from pull requests that are not yet in master branch.

staal54a, 2024-06-24:
I cleaned and applied those to 2.8.0-beta1 and it's still erroring out. See the following screenshot:
[image]

schiele, 2024-06-26:
> C++ includes are hard. We compile the software on 3 different platform with different compilers and see no issues, but still this happens quite often for other compilers/compiler versions.

From the pattern I have seen so far I suspect that you build internally with precompiled header support, which often hides missing header dependencies. CMake turns that on by default for some generators. If that is the case, consider doing regular builds without precompiled header support to catch those errors before releasing.