prusa3d/PrusaSlicer #12862: Make adding new Filament profiles easier

dwbmb, 2024-06-19:
I've been using PrusaSlice for year and a half and today I realised I was unable to add new Filament to choose from - at least not where I expected such an option to be - on the Filaments tab.

[image]

There are listed only filaments already set-up previously, no obvious option to add another one. I found it to be hidden in in Plater tab - Filament-Add/Remove filament option at the bottom of the list. Which opens the whole configuration wizard by the way.

[image]

Si I would like option to add new Filament profile:

1) With easier process, without going through configuration wizard, just Filaments selection
2) To access it from a Filament tab, ideally by dedicated button and not hidden somewhere at the bottom of the drop-down list.

Or at least adding this option to the bottom of that drop-down list on the Filaments tab would be a great first step.

u89djt, 2024-06-19:
While developers think about whether it's worth adding buttons, you can just change the parameters in the filaments tab in your first screenshot to what you need, then hit the save button and give the new preset a name in the dialog that pops up.
[image]

dwbmb, 2024-06-19:
I was aiming at using pre-built profiles in PrusaSlicer more than creating new profiles but thanks.

closed 2026-09-29 by prusaslicer-bot[bot]