prusa3d/PrusaSlicer #12865: prusaslicer-avatar-1539752056057364272.png

semich, 2024-06-19:
### Description of the bug

Since using the PrusaSlicer-2.8.0-alpha5+win64-202406142204, I get the following error message, when I start the Slice:
19:56:11: This is not a PNG file.
19:56:11: Failed to load image from file "C:\Users\semich\AppData\Local\Temp\prusaslicer-avatar-1539752056057364272.png".
19:56:11: This is not a PNG file.
19:56:11: Failed to load image from file "C:\Users\semich\AppData\Local\Temp\prusaslicer-avatar-1539752056057364272.png".

### Project file & How to reproduce

Start slicer - error appears.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0-alpha5+win64-202406142204

### Operating system

Windows 11

### Printer model

MK4

Jan-Soustruznik, 2024-06-24:
Hello, @semich , could you share your avatar, please? thank you

semich, 2024-06-24:
Hmmm I didn't know I had an avatar :) will check

semich, 2024-06-24:
I setup a new one on my online profile and it is solved - thank you!

closed 2024-06-24 by semich

mskrohn, 2024-06-24:
My google photo defaulted as my profile picture and caused this error.  Going to the Prusa website and adding a profile picture to my account fixed this.

jackthese, 2024-06-25:
mskrohn..."My google photo defaulted as my profile picture and caused this error. Going to the Prusa website and adding a profile picture to my account fixed this."
So, I'm still getting this error message when I open PrusaSlicer 2.8 beta, and my Prusa account has my picture. What, exactly, is the fix? Love the product using PrusaConnect. I'm connected to my Mini + on my LAN, not WiFi, incidentally.

mskrohn, 2024-06-25:
> mskrohn..."My google photo defaulted as my profile picture and caused this error. Going to the Prusa website and adding a profile picture to my account fixed this." So, I'm still getting this error [...]

Mine had a picture too, but it was from my Google log in. Upload a new picture to your Prusa account!

Jan-Soustruznik, 2024-06-25:
https://github.com/prusa3d/PrusaSlicer/issues/12860

lukasmatena, 2024-06-28:
Thanks. We managed to find and fix the problem. It was not in PrusaSlicer, but on the server side. The fix was already deployed, so this issue should no longer show in any PrusaSlicer version.

Avatars from Prusa Account showed correctly, the problem appeared when avatar from Google account was used.

Can you guys confirm that the issue really disappeared?

semich, 2024-06-28:
I don't remember I how setup that picture I had before, I did it like 5 or 6 years ago... 🙁

jackthese, 2024-06-28:
Just tried it in 2.8.0 and it seems OK to me, thanks

Trohorm, 2025-02-19:
I'm 2.9.0 and the issue still happens when i start the slicer or when I'm working in it. I was not doing any printing or slicing in a year so i don't really know. Just wanna let you know.
This is the message i get: 
20:28:00: This is not a PNG file.
20:28:00: Failed to load image from file "C:\Users\thoma\AppData\Local\Temp\prusaslicer-avatar-5560774625658756674.png".
20:28:00: This is not a PNG file.
20:28:00: Failed to load image from file "C:\Users\thoma\AppData\Local\Temp\prusaslicer-avatar-5560774625658756674.png".

Tho the second i changed my profile picture the issue stopped. Idk i might be useless i just wanna let ppl know the issue is still there even in the newest version at the time.