prusa3d/PrusaSlicer #12870: Tests and Calibrations

Tony3DPrinting, 2024-06-20:
Why don't you implement testing and calibration?
It would be the best, it would be a software complete with everything.

u89djt, 2024-06-20:
Fellow user here, not affiliated with Prusa. There's not a lot can be said except that you can do whatever calibration and testing you want to do. https://help.prusa3d.com/category/calibration_199
There are functions some printers can do and others can't because of the presence or absence of appropriate hardware. For example my XL can do input shaping and phase stepping, but my MK3S can't because it doesn't have an accelerometer, and the servos and mainboard are older models.
When you've read the articles and had a go at the methods provided, if you have a specific question about something you can't achieve, someone's likely to try to help.

rtyr, 2024-06-20:
This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6168 and possibly others, we are considering something like that.

closed 2024-06-20 by rtyr