prusa3d/PrusaSlicer #12871: Duplicate File Name Upload to Connect Fails

Luke240, 2024-06-20:
### Description of the bug

When uploading a file from slicer to connect, if there is already an existing file with the same name, the upload fails rather than trying to offer a name change alternative, or auto incrementing. 

### Project file & How to reproduce

Upload a print to connect, then immediately upload again. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.8.0-alpha5+MacOS-arm64

### Operating system

MacOS 10.15

### Printer model

Prusa XL

kubispe1, 2024-06-20:
Thanks, this is know issue, but not resolved yet. 
CONNECT-2531

ngolf, 2024-10-07:
I have a similar issue, except I don't get the dialog box.

This causes the upload (and print start)to fail silently.

cc2224cody, 2024-11-16:
Same issue as ngolf here. With Prusa Link it works fine. After changing to Connect (had problems with the profiles), if I send a print to the MK4S via Prusaslicer & Connect and then resent it (e.g. after changing the position) while the printer is idle, it just does nothing. 
Temp fix: change name during upload dialog, then it starts the print just fine.

ngolf, 2024-11-17:
I struggle to see a case where overwriting the existing file is not the better option.

closed 2026-09-29 by prusaslicer-bot[bot]