prusa3d/PrusaSlicer #12874: Ctrl-P does not open preferences in 2.8.0-alpha5

DoubleStrike, 2024-06-20:
### Description of the bug

Pressing Ctrl-P should open the Preferences dialog.  It does not.

### Project file & How to reproduce

Open a fresh copy of PS.  Press Ctrl-P.  Marvel at the beauty of nothing happening.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0-alpha5

### Operating system

Windows 10 22H2

### Printer model

XL 5-toolhead with input shaper

Jan-Soustruznik, 2024-06-24:
Hello, @DoubleStrike , thank you for your report. We will try to fix in some next releases. Have a nice day 

Issue created:  SPE-2372

Jan-Soustruznik, 2024-06-25:
Fixed in https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-rc1 , closed as resolved. Thank you, have a nice day

closed 2024-06-25 by Jan-Soustruznik