prusa3d/PrusaSlicer #12876: Option to set up 1st Layer flow factor different like the temperature for Color Fab LW-PLA-HT

BernhardFe, 2024-06-21:
I'm always l little frustrated when I need to edit G-Code for a functionality or need to user Simplify3D instead of Prusa slicr because of missing function.

For printing planes I like to use ColorFab LW-PLA-HT filament. But this requires for the first layer a temperature of 200C an a flow factor of 100%. But changing the second layer to 240C the flow factor need to be reduced in my case to 52% because the filament blows upon on the nozzle. 

For this I like to have the flow factor for the 1st layer different together with the temperature as for the rest of the model.
All can be changed for the first layer in Prusa Slicr but not the flow factor what is really mandatory here.

Edit G-Code ore use Simplify3D is an option, but I don't like it very much.

Anyhow, a profile for this special filament would be great for people need to print lightweight models.
[Color Fab LW-PLA-HT](https://colorfabb.com/filaments/materials/pla-filaments/lw-pla-ht)

BR Bernhard

u89djt, 2024-06-21:
(fellow user reposting a workaround)
Here's someone posting custom gcode to do this:
https://forum.prusa3d.com/forum/prusaslicer/first-layer-extrusion-multiplier/#post-694673
It goes here
[image]

BernhardFe, 2024-06-21:
Ok, Thank you very much! This would be a short term solution. I will use it.
But I think a field would be very useful and easier even if more and more people will have this question. :-)

closed 2026-09-29 by prusaslicer-bot[bot]