prusa3d/PrusaSlicer #12877: Color Change Added After Moving Slider

dunkmann00, 2024-06-21:
### Description of the bug

When you drag the slider and then move it back to the starting position, if the cursor is inside the slider button when releasing the mouse button, it will add a color change. If I already dragged the slider, I would expect nothing to happen when I release the mouse button.

This might seem unlikely to happen, but I had it happen to me a bunch of times. At that point I tried to pay attention to what I was doing to make it happen, because at first it seemed random, but as it turns out, it wasn't.

### Project file & How to reproduce

https://github.com/prusa3d/PrusaSlicer/assets/11963090/46d349cc-81f5-477e-b06e-d05220510c35

[SliderColorChangeBugReport.3mf.zip](https://github.com/user-attachments/files/15934406/SliderColorChangeBugReport.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.8.0-beta1+MacOS-universal-202406212048

### Operating system

macOS 14.4

### Printer model

MK4

Jan-Soustruznik, 2024-06-24:
hello, @dunkmann00 , thank you for report, have a nice day

issue created SPE-2371

lukasmatena, 2024-06-25:
This should be fixed in [2.8.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-rc1). Closing.

closed 2024-06-25 by lukasmatena